OpenTracksApp/OpenTracks #10: Present elevation gain (via barometer)

dennisguse, 2019-08-09:
GPS data is usually to imprecise to record the elevation gain (therefore it was disabled in OpenTracks).
Use the barometer to record the elevation gain.

dennisguse, 2019-12-11:
I did some research on this topic, especially the sensor fusion part.
Most notably, it is important to notice that both sensors are different:

* GPS (high variance; high accuracy)
  * provides absolute altitude (no drift)
  * absolute accuracy cannot get below 5m (more often worse)

* Barometer (low variance; low accuracy)
  * absolute accuracy can be determined using formula (requires pressure at sea level + temperature): see http://www.hills-database.co.uk/altim.html
  * drift: is affected by weather (slow changes over time; i.e., morning vs evening)
  * 
An approach for sensor fusion is described here: https://users.ece.cmu.edu/~franzf/papers/baroaltitude.pdf

dennisguse, 2020-01-29:
Another option is store the barometric (relative) ascend/descend separately for each TrackPoint in addition to the absolute GPS value.
Then the elevation gain/decrease can be easily reported and actually no sensor fusion takes place. 
Also varying varying/different intervals of the sensors can then be ignored.
However, exporting/importing the data to KML/GPX might be a problem...

dennisguse, 2020-04-11:
Current implementation approach: store the elevation gain in the `TrackStatistics`.
This does not solve the import/export problem.

dennisguse, 2020-06-21:
@rgmf Thanks for your test and suggestion.
I now added two more "enhancements" (more like try and error ><):
1. elevation changes are now limited to 5m (using the defined constant).
If an elevation change is detected, the sensor value is actually computed based (implemented invert of SensorManager.getAltitude()).
2. Exponential smoothing is applied 
(I was to lazy to implement a moving average...)

I will do some really basic (flatland) test on Tuesday.

dennisguse, 2020-08-08:
I did a 120km test today and got 250m gain while the Etrex 30 reported 450m.
But it was flat land (really no mountains), so I guess it is was okay :)
And the trip took 9hours, so weather changes are included (normal once during the day, bright sunshine without rain).

rgmf, 2020-08-08:
> I did a 120km test today and got 250m gain while the Etrex 30 reported 450m.
> But it was flat land (really no mountains), so I guess it is was okay :)
> And the trip took 9hours, so weather changes are included (normal once during the day, bright sunshine without rain).

Garmin will count every meter and OpenTracks only count if you climb at least 3 meters so in flat land you can lost a lot of gain information. This can be the explanation.

closed 2020-09-06 by dennisguse