OpenTracksApp/OpenTracks #14: Add new activity types (incl. icons)

dennisguse, 2019-08-09:
[ ] MTB
[ ] eScooter
[ ] Motorcycle
[ ] Skateboard
[ ] Sailing
[ ] Inlineskater

dennisguse, 2020-01-07:
A friend of mine takes care of this - this might take some time.

dennisguse, 2020-07-14:
Currently, available icons.

[image]

mbblp, 2020-07-20:
[icons.zip](https://github.com/OpenTracksApp/OpenTracks/files/4947606/icons.zip)
these are the 4 icons for testing

dennisguse, 2020-07-27:
@mbblp I just had the time to check the images.
Nice work!

While trying ot import I get the following error message: `WARNING @ line 229: Scaling of the stroke width is ignored`.
Some infos to fix this can be found here: https://stackoverflow.com/questions/36453520/inconsistence-between-svg-file-and-android-studio-vector-asset

Do you know how to fix this?

pstorch, 2020-08-08:
All listed activity types are added. If we need more, we can open a new issue.

closed 2020-08-08 by pstorch