OpenTracksApp/OpenTracks #15: Rework stats_layout

dennisguse, 2019-08-09:
The stats_layout (used by TrackDetailActivity) has highly nested layout.
This should be simplified to make maintenance easier as well as the layout faster.

dennisguse, 2020-05-13:
Partly done in #229.

rgmf, 2020-05-22:
> Using `GridLayout` would an option.

What do you think about ConstraintLayout? The highly nested layout would disappear. I could follow #229 trying to convert the nested LinearLayouts to ConstraintLayout.

dennisguse, 2020-05-22:
@rgmf +1
I just don't like that adds a dependency - but that is a personal opinion.

rgmf, 2020-05-22:
@dennisguse then a GridLayout will be the best option.

dennisguse, 2020-05-22:
@rgmf Don't worry, I will get over it.
You can just give it a try if you want.
I mean the ConstraintLayout might also get rid of these empty spaces that under some settings occur.
So, I guess it is worth a shot ;)

rgmf, 2020-05-23:
Done in #229

commit fb31697a (2020-05-23): Recorded/Recording nested layouts converted to constraint layout. Fixes #15.
commit 12e6f030 (2020-05-25): Recorded/Recording nested layouts converted to constraint layout. Fixes #15.
closed 2020-05-25 by rgmf