OpenTracksApp/OpenTracks #24: Heartrate bluetooth sensor: show battery info

dennisguse, 2019-08-11:


gordonel, 2021-05-22:
**Is your feature request related to a problem? Please describe.**
I'd like to be notified if it's time to charge my devices or replace the batteries in advance.

**Describe the solution you'd like**
When starting an activity, OpenTracks should read the battery level if available and notify the user if it falls below a user-specified level. Also, the app can provide convenient readouts in the notifications drawer as extra info

**Describe alternatives you've considered**
reading battery levels through nRF connect before a workout

dennisguse, 2022-05-12:
Parsing is already there; we "just" miss the UI.
Probably a notification would be sufficient in case of low battery (=<15%)?

edern-appere, 2022-05-13:
Yes, a notification would be great !

Would it be possible to show the notification "at any time" ? For example, when starting the application, if the battery of sensor is already low (=< 15%) displaying the notification would inform that the sensor won't last very long before begining a road trip...