OpenTracksApp/OpenTracks #27: Setup Translation management systems (TMS)

dennisguse, 2019-08-11:
Option would be to use [translate5](https://translate5.net/)?

dennisguse, 2019-11-04:
Just setup Transifex - might be sufficient.

dennisguse, 2020-04-19:
We are now migrating to crowdin: https://crowdin.com/project/opentracks

Todo:
[ ] setup git integration (files are not correctly imported)
[ ] add link to everywhere needed (README.md / app?)

pstorch, 2020-04-19:
Regarding the folder pattern, which are wrong in the PR. When you download it manually it seems to be right:

[image]

pstorch, 2020-04-19:
I got a new PR with still a wrong file Pattern with `de-rDE`, I solved it now via Export / Language Mapping:

[image]

dennisguse, 2020-04-19:
I guess you are using android-codes, right? For OpenTracks I tried yesterday two-letter-codes.
However, somehow no automatic import is taking place....

pstorch, 2020-04-19:
Yes, I also tried the two-letter-codes, but then I got a Exclamation Mark at the syncing, because I have a language with a variant (Portuguese Brazilian) which is not compatible with two-letter-codes.

dennisguse, 2020-04-19:
OK, but syncing worked? Because in the OpenTracks project syncing is not kicking in...
OpenTracks has French, Chinese and Portuguese that are the only android-codes.
I was thinking about using a custom language mapping.
Also we have support for Spanish South America and English (not yet figured out how to do this in Crowd in).

pstorch, 2020-04-19:
I also had some issues with the syncing. Suddenly it was working again. After doing some more translations and forcing the build.
English is the default language for OpenTracks, right? So there is no translation necessary. You need to tick it off.

pstorch, 2020-04-19:
I would also set "Skip untranslated strings" otherwise each language file contains all the english source strings as duplicates. Also "Skip untranslated files" should be set, I guess.

[image]

Can you also try to approve all translations so far, because anyway they are in the App right now. Maybe that helps triggering the build and syncing.

dennisguse, 2020-04-19:
Thanks, will try this.
English US is default.
English UK is the other.

dennisguse, 2020-04-19:
Re-enabled transifex - hope that works better.
Removed crowdin commits from master.

dennisguse, 2020-04-20:
Here is the transifex url: https://www.transifex.com/opentracks

For Transifex, the Github integration is activated - this _should_ automatically pull changes from the repo.
Push from Transifex to Github happens if 100% are translated (opens a PR).

For crowdin, I failed to setup the sync from Github - somehow the files would need to be uploaded manually(?) and changes in the repo are not reflected.
If somebody has an idea/experience, please let me know.

dennisguse, 2020-04-28:
Transifex is up and running.

Workflow requires that all changes need to be reviewed and by default only languages are added via PR if 100% is translated.
Otherwise, we can trigger the process manually under settings/integration.

commit bf3b529f (2019-11-07): Add link to Transifex for localization.

Fixes #27.
closed 2020-04-28 by dennisguse