OpenTracksApp/OpenTracks #40: [FR] GadgetBridge data fetching

ildar, 2019-09-20:
As the app supports heart rate then it's easier getting it from a fitness tracker. And subj is one open platform which can provide this info from a number of fitness trackers. Gpx available from some trackers too.

dennisguse, 2019-10-13:
Any suggestions on how to implement this?

dennisguse, 2019-11-02:
I just thought about this: OpenTracks stores the data for each captures GPS position (i.e., one heart rate entry per position). So importing heart rate alone is technically not an option. In addition, the connection to the heart rate sensor (Bluetooth LE) only can read the actual heart rate and no historic data from the device.

The easiest way to import data would be to provide them as GPX/KML (and include the heart rate as additional data).
If you think this might be worth the effort please open a new issue.

closed 2019-11-02 by dennisguse