OpenTracksApp/OpenTracks #49: Crash when opening Settings activity

vanous, 2019-10-21:
Thanks for this great app!

I have searched open issues but didn't find this one, sorry if i have missed it.

For some reason, after a while, settings stopped working - when i try to open settings screen, app crashes.

Version V3.2.1, from F-Droid.
Samsung S7 Edge, Android 8.

Below is adb shell logcat

thank you

```
10-21 19:04:03.697  3671 14055 D ActivityManager: isScaleDownAnimationEnabled() : false
10-21 19:04:03.698  3671 14055 V WindowManager: rotationForOrientationLw(orient=-1, last=0); user=0  sensorRotation=-1 mLidState=-1 mDockMode=0 mHdmiPlugged=false
10-21 19:04:03.702  3671 14055 D PersonaManagerService: Current focused persona service handled id set to : 0
10-21 19:04:03.702  3671 14055 D InputDispatcher: Focused application set to: xxxx
10-21 19:04:03.705  3671 14055 D InputDispatcher: Focus left window: 18097
10-21 19:04:03.707  3671  3962 D GameManagerService: handleForegroundChange(). pkgName: de.dennisguse.opentracks, clsName: de.dennisguse.opentracks.settings.SettingsActivity,FgActivityName:de.dennisguse.opentracks/.settings.SettingsActivity
10-21 19:04:03.707  3671  3962 D GameManagerService:   handleForegroundChange(). same package with Prev KILL_YOURSELF. ignore
10-21 19:04:03.708  3671 14055 D MARsPolicyManager: onPackageResumedFG pkgName = de.dennisguse.opentracks, userId = 0
10-21 19:04:03.716 18097 18097 D ViewRootImpl@4a19370[TrackDetailActivity]: MSG_WINDOW_FOCUS_CHANGED 1
10-21 19:04:03.717  3671 14055 D InputMethodManagerService: startInputOrWindowGainedFocus : windowGainedFocus 
10-21 19:04:03.717  3671 14055 D InputMethodManagerService: windowGainedFocus: destinationUserId (getCallingUid) = 0
10-21 19:04:03.717  3671 14055 D InputMethodManagerService: windowGainedFocus: currentUserId=0
10-21 19:04:03.717  3671 14055 D InputMethodManagerService: windowGainedFocus: mCurrentFocusedUserId=0
10-21 19:04:03.718  3671 14055 D InputMethodManagerService: windowGainedFocus : lock - currentUserId = 0 destinationUserId = 0
10-21 19:04:03.718  3671 14055 D InputMethodManagerService: windowGainedFocus : mCurrentFocusedUserId - 0 and mSecureKeypadEnabled - false, userSwitched - false
10-21 19:04:03.718  3671 14055 W InputMethodManagerService: Focus gain on non-focused client com.android.internal.view.IInputMethodClient$Stub$Proxy@cba9bf4 (uid=10041 pid=18097)
10-21 19:04:03.718  3671 14055 W InputMethodManagerService: calledFromValidUser : true, (mCurFocusedWindow == windowToken) : true, (attribute != null) : false
10-21 19:04:03.718  3671 14055 W InputMethodManagerService: (mCurClient != cs) : false, mCurFocusedWindow : android.os.BinderProxy@4c83c2, windowToken : android.os.BinderProxy@4c83c2
10-21 19:04:03.718 18097 18097 E ViewRootImpl: sendUserActionEvent() returned.
10-21 19:04:03.730 18097 18097 D AndroidRuntime: Shutting down VM
10-21 19:04:03.731 18097 18097 E AndroidRuntime: FATAL EXCEPTION: main
10-21 19:04:03.731 18097 18097 E AndroidRuntime: Process: de.dennisguse.opentracks, PID: 18097
10-21 19:04:03.731 18097 18097 E AndroidRuntime: java.lang.RuntimeException: Unable to start activity ComponentInfo{de.dennisguse.opentracks/de.dennisguse.opentracks.settings.SettingsActivity}: java.lang.ClassCastException: java.lang.Integer cannot be cast to java.lang.String
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2957)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3032)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at android.app.ActivityThread.-wrap11(Unknown Source:0)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1696)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:105)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:164)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:6944)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:327)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1374)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: Caused by: java.lang.ClassCastException: java.lang.Integer cannot be cast to java.lang.String
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at android.app.SharedPreferencesImpl.getString(SharedPreferencesImpl.java:255)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at androidx.preference.Preference.getPersistedString(Preference.java:1686)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at androidx.preference.ListPreference.onSetInitialValue(ListPreference.java:255)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at androidx.preference.Preference.onSetInitialValue(Preference.java:1614)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at androidx.preference.Preference.dispatchSetInitialValue(Preference.java:1587)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at androidx.preference.Preference.onAttachedToHierarchy(Preference.java:1311)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at androidx.preference.Preference.onAttachedToHierarchy(Preference.java:1326)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at androidx.preference.PreferenceGroup.addPreference(PreferenceGroup.java:249)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at androidx.preference.PreferenceGroup.addItemFromInflater(PreferenceGroup.java:170)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at androidx.preference.PreferenceInflater.rInflate(PreferenceInflater.java:345)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at androidx.preference.PreferenceInflater.rInflate(PreferenceInflater.java:346)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at androidx.preference.PreferenceInflater.inflate(PreferenceInflater.java:157)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at androidx.preference.PreferenceInflater.inflate(PreferenceInflater.java:109)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at androidx.preference.PreferenceManager.inflateFromResource(PreferenceManager.java:216)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at androidx.preference.PreferenceFragment.addPreferencesFromResource(PreferenceFragment.java:386)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at de.dennisguse.opentracks.settings.SettingsActivity$PrefsFragment.onCreatePreferences(SettingsActivity.java:65)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at androidx.preference.PreferenceFragment.onCreate(PreferenceFragment.java:167)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at android.app.Fragment.performCreate(Fragment.java:2593)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1234)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at android.app.FragmentManagerImpl.addAddedFragments(FragmentManager.java:2421)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at android.app.FragmentManagerImpl.executeOpsTogether(FragmentManager.java:2200)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at android.app.FragmentManagerImpl.removeRedundantOperationsAndExecute(FragmentManager.java:2154)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at android.app.FragmentManagerImpl.execPendingActions(FragmentManager.java:2055)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at android.app.FragmentManagerImpl.dispatchMoveToState(FragmentManager.java:3050)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at android.app.FragmentManagerImpl.dispatchActivityCreated(FragmentManager.java:2997)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at android.app.FragmentController.dispatchActivityCreated(FragmentController.java:179)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at android.app.Activity.performCreateCommon(Activity.java:7177)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at android.app.Activity.performCreate(Activity.java:7185)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1220)
10-21 19:04:03.731 18097 18097 E AndroidRuntime: 	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2910)
[... cut ...]
10-21 19:04:03.738  3671 14055 D Debug   : !@DumpState : debug level:0x4f4c
10-21 19:04:03.738  3671 14055 D Debug   : !@Dumpstate : Finally, system will skip dumpstate
10-21 19:04:03.738  3671 14055 W ActivityManager: crash : de.dennisguse.opentracks,0
10-21 19:04:03.739  3671 14055 W ActivityManager:   Force finishing activity de.dennisguse.opentracks/.settings.SettingsActivity
10-21 19:04:03.748  3671 14055 W ActivityManager:   Force finishing activity de.dennisguse.opentracks/.TrackDetailActivity
```

dennisguse, 2019-10-21:
Thanks for some reason one of the SharedPreferences seems to be stored as an Integer while it should be a String. 
The only thing that you can try is to delete all app data via Android settings (which removes all tracks and also all settings).
I am not entirely sure, which preference it was - could you try to just use the app and check if it occurs again?

vanous, 2019-10-22:
Cool, sure, erasing helps. I can also self compile, if you want me to test something.

vanous, 2019-10-22:
Hmm, as one would expect ;) , i did not re-create it. I use Amazfit Bip as BLE HR sensor, so i tried using it too, changing settings i might have changed, changing all other settings around... to no avail.... issue not replicated.

dennisguse, 2019-10-22:
Good to hear that also other sensors are working properly.
We can leave the issue open until end of the year and see if something related comes up again.

@vanous Thanks for investigating it.

dennisguse, 2019-10-31:
It seems to be a bug in the tests.
I just found a way to reproduce this issue:
1. Install app (only once); check that Settings can be opened
2. Run Tests on this device (via AndroidStudio)
3. Open app and go to Settings: crash

When clearing the app's storage via Android Settings, everything is fine again.

vanous, 2019-10-31:
This is great. I have replicated again but wanted to try to get app data. Unfortunately on my rooted phone I didn't manage to pair my watch (wasn't sure if that's needed or not) and on non rooted didn't find time to try to get app data with adb backup. ..

vanous, 2019-10-31:
The adb backup and then restore never works for me...

Anything i can try to help with now? How do i run the test?

dennisguse, 2019-10-31:
@vanous You can just clone the repository and open it in AndroidStudio or by using `./gradlew connectedCheck`.

I just followed up the finding that the issue is (also) triggered by the tests.
The issue occurs for `TrackRecordingServiceTest` and `TrackRecordingServiceTestLooper` due to this function:

https://github.com/dennisguse/OpenTracks/blob/afd4a59622bab96630290f4414b1b8f4c95ea0a1/src/androidTest/java/de/dennisguse/opentracks/services/TrackRecordingServiceTest.java#L80

due to 
https://github.com/dennisguse/OpenTracks/blob/afd4a59622bab96630290f4414b1b8f4c95ea0a1/src/androidTest/java/de/dennisguse/opentracks/services/TrackRecordingServiceTest.java#L82

My guess is that the `ListPreference` require a `String` in the `SharedPreferences` while an `int` is found...
A similar issue might occur in the non-testing code (i.e., modifying settings from the code while being changable via the Settings activity).
I believe, we would need to look into the use of `PreferenceUtils.set*`...

dennisguse, 2019-10-31:
And I have found another issue: https://github.com/dennisguse/OpenTracks/blob/a83eb0b983a4911155cb12c9af2b6c077a783dc2/src/main/java/de/dennisguse/opentracks/fragments/FrequencyDialogFragment.java#L67

This dialog allows to change the frequency of voice commands while being in the `TrackDetailActivity` and recording.
And it should set a `String` instead of an `int`.

@vanous could it be that you used this function?
My fix now would be to remove this completely.
What do you think?

vanous, 2019-11-01:
I am sure i used this the first time i created this issue. Strange thing was, that when i tried to re-create it, the issue didn't come up, or at least not right away. Tracking was not enabled at that moment, so the system did not have any reason to keep the app running, so regular restart should kill/reopen it. I have a feeling that that didn't happen and perhaps the issue only surfaced after the app got garbage collected, closed and then reopened in the future...

I think it would be pity to remove this functionality, as it is useful to set it to say 1 minute etc.

 As for direct access to preferences... normally, i do not observe issue in other projects... but i didn't dig deeper here. Maybe the confusion is in the array lists and the need to be more explicit.

closed 2019-11-01 by dennisguse

dennisguse, 2019-11-02:
@vanous Thanks for taking the time reporting the issue.
If you have any other issues or suggestions/ideas, please let me know.

Release v3.2.4 contains the fix and should be in Fdroid in some days.

commit d53a7156 (2019-11-01): Bugfix: removed custom dialogs to set voice/split frequency preferences from the context menu directly.
Fixed #49.