OpenTracksApp/OpenTracks #56: No GPS data

r4taw2mg, 2019-11-03:
Hi,

The installation on my Sony Z3 does not work. The app is not getting any gps data. GPS is turned on, location permission has been granted. Other apps (e.g. osmAnd) do work. Rooted D6603 running Android 6.0.1.

dennisguse, 2019-11-03:
Thanks for the report.
Just a quick guess: Did you change the _GPS accuracy_ setting?
And if not could you set it to a larger value?
I just realized -really 20min ago- that this setting might be misleading; basically all GPS coordinates are filtered before actually being processed if their accuracy is larger than the threshold.

If this does not solve the issue:
Could you provide some more information:
* which version of OpenTracks do you use (can be found in the Settings)?
* what OS do you run (e.g., Sony original or LineageOS)?

And would be possible to get a logcat output (when starting a recording)?

r4taw2mg, 2019-11-04:
Hi, 

thanks for your assistance. I didn't change any of the GPS settings and it was Sony firmware. But I made an update and it worked with the new version. (I was one release behind).

I am closing the ticket.

Cheers
Frank

closed 2019-11-04 by r4taw2mg

dennisguse, 2019-11-04:
Interesting that the bug fixed itself as the new version 3.2.3 as well as 3.2.4 did not contain any changes that are related to the GPS.
So, if you encounter the issue again - please reopen this issue.

If you have any other feedback and see room for improvement - let me know.

r4taw2mg, 2019-11-05:
I understand. Nonetheless that was my experience.

I don't get any notification on my lock screen. Some permissions missing?