OpenTracksApp/OpenTracks #61: Android 10: App not capturing GPS tracks while display is off

vanitasvitae, 2019-11-04:
Hi!
It appears that OpenTracks is not capturing GPS information while the app is in background or the screen is off. I expect this to be caused by Android 10 only allowing access to location data while the app is actively being used. The permission menu only offers the options to disallow access, or allow access while app is being used.

I'm not sure if there is a way around this by requesting an exception from the system.

dennisguse, 2019-11-05:
I just pushed the change to master.
Would you be able to test this, so it can be released?

vanitasvitae, 2019-11-05:
I can try building the app tomorrow evening :)
Thank you for the quick response and the fix :+1:

dennisguse, 2019-11-05:
Please test using: https://github.com/dennisguse/OpenTracks/commit/53572406e9cd1796f0b418b16903addad1f1adca

In the following commits I messed something up while refactoring the preferences.
This leads to the situation that no recording can be started.
I hope I have time to fix it tomorrow.

dennisguse, 2019-11-06:
I fixed the issues that I introduced while refactoring.
The are already in the master branch.
https://github.com/dennisguse/OpenTracks/tree/d3f5080e7311d49386539bc820509546839f0aac

vanitasvitae, 2019-11-06:
I briefly tested the app from the current master branch and can confirm that the issue appears to be fixed. I didn't have a chance yet to really test on my device (will do that tomorrow), but at least the emulator correctly recorded a simulated route :)

Nice!

vanitasvitae, 2019-11-07:
Just tested over an extended period on my phone and confirm the issue to be fixed. Great job :)

dennisguse, 2019-11-07:
If you have time, you could briefly check if the settings are working as you expect. I did some nasty refactoring and is not yet well tested and I want to avoid delivering a broken update to Fdroid...

vanitasvitae, 2019-11-07:
What specific setting do you refer to? I should mention that I'm a new user, so I'm not very familiar with the settings and how they are supposed to work (as in I wouldn't be able to spot behavior that changed in the new version).

dennisguse, 2019-11-07:
About testing the settings: just open and check if you notice something weird ;)

About the MiBand2: I don't know if supports streaming the heart rate, but I doubt it as stores the data locally and later syncs it.
And this functionality is not available in OpenTracks.
However, #58 is related as the MiBand should not have been selectable.

dennisguse, 2019-11-07:
And we need to leave this issue open until the next release is finished as somebody else might have the same issue.

vanitasvitae, 2019-11-07:
> And we need to leave this issue open until the next release is finished as somebody else might have the same issue.

Ah makes sense.

The only "bug" I found in settings is, that changing the preferred metric (second entry from the top) doesn't update the description text when being changed.

dennisguse, 2019-11-09:
I crafted version 3.2.5 yesterday, which includes the GPS fix as well as fixes the summary issue.

commit 53572406 (2019-11-05): Mark foreground service to use location and connection to other devices, so it can run properly on Android 10+ when app is in the background.
Might fix #61.
closed 2019-11-09 by dennisguse