OpenTracksApp/OpenTracks #63: heart rate technology (ant+)

Bebarr, 2019-11-05:
Hi! Thanks for your app. It's really cool.
Is it possible, with upgrades, to use my old heart rate belt with the ant+ technology?

dennisguse, 2019-11-05:
Sadly no (at least not for the time being).

In the original version of MyTracks (predecessor of OpenTracks created by Google) ANT+ was supported.
However, this required an additional dependencies (I forget which ones were used) as Android does not support Ant+ by default.
As the dependencies were really old (pre  2013) and maintaining the integration would have been quite some amount of work I removed it.

I also removed Bluetooth classic and added Bluetooth LE as only supported protocol (which is more or less state of the art).
The major advantage is maintainability...

So if somebody has time to add ANT+ nicely - go ahead....

closed 2019-11-05 by dennisguse

Bebarr, 2019-11-21:
> So if somebody has time to add ANT+ nicely - go ahead....

Thanks for your answer. I understand ;) unfortunately I'm just a user. I don't know how to code..