OpenTracksApp/OpenTracks #67: Crash when selecting Parameters right after start up

serpico, 2019-11-09:
**Describe the bug**
When tapping on the vertical three dots on the upper right corner and selecting Parameters right after starting the app it crashes

**To Reproduce**
1. Launch the app
2. Tap on the vertical three dots on the upper right corner
3. Select Parameters
4. The app crash/close

[opentracks.txt.](https://github.com/dennisguse/OpenTracks/files/3826971/opentracks.txt)

**Technical information**
 - Device: Samsung SM-G361F
 - OS: original ROM Android 5.1.1
 - Version [e.g. v3.2.3]

dennisguse, 2019-11-09:
This bug was fixed in v3.2.5, which I released yesterday and should be in Fdroid in the next days. If the settings are invalid (and thus the Settings activity would crash) all settings are reset to default. 
The underlying issue was probably the shortcut while recording a track and setting the voice frequency. 
Alternatively, you could delete all app data (which also deletes all tracks).
If the error persists on v3.2.5 please let me know.

serpico, 2019-11-09:
Thanks @dennisguse I'll make sure to update when v3.2.5 will hit Frdoid and let you know so you can close the issue.

dennisguse, 2019-11-09:
Just close the issue if everything works again ;

serpico, 2019-11-09:
will do

dennisguse, 2019-11-11:
v3.2.5 just got available on the F-Droid store.
@serpico  If you update, you can go to the settings and a message should be shown that an error occurred and the settings were reset.
Then everything should work fine.

closed 2019-11-12 by serpico