OpenTracksApp/OpenTracks #69: Voice announce covered by music when listening during run/ride

serpico, 2019-11-12:
**Is your feature request related to a problem? Please describe.**
When Voice Frequency is enabled and the user simultaneously play some music ( using VLC ), the music is not paused/muted when the announce is made/voiced, but appear to be "voiced over" making it difficult to understand what is announced.

**Describe the solution you'd like**
Some other app on f-droid ( Tasks, OsmAnd ) the music is paused or muted so announce can be heard clearly

Thanks

dennisguse, 2019-11-12:
I actually thought this functionality was in place (but I am not sure).
Could you check what happens if you have call in parallel?
Because for this case, I saw the implementation (although never tested it).

dennisguse, 2019-11-13:
@serpico I just tested it with v3.2.5 (from FDroid store) and it work as expected (music is silenced but playing continuous).
As music player I used Odyssey: https://f-droid.org/en/packages/org.gateshipone.odyssey/

Can you try to reproduce this issue with another player than VLC?

serpico, 2019-11-15:
tl;dr : it mostly depends on the volume of the tracks/song I'm listening to, some tracks I had the volume increased using easyMP3Gain, they're the troublesome ones, as lowering the media volume via the button on the side of the phone, will also lower the volume of the announcement by OpenTracks, with some other songs the volume of the song is similar to the volume of the announcement in OpenTrack so when the media player output is lowered it's  easier to hear the OT announcement, still I would increase OT's volume a bit if I could ( especially during ride ).

I did a few tests and realized in OpenTracks the volume of the media player output was only lowered not muted entirely, ( muted as no sound but the announcement, similar to Tasks or OsmAnd ).

I think I noticed a slight difference between Odyssey and VLC, to the benefit of Odyssey, but it's hard to tell because it change on a song to song basis as described above, in both media player volume of the media player is only lowered not silenced though.

When I poked around the sound settings in the parameter's menu of the phone ( four separate volume settings : Ringtone ; Media ; Notifications ; System ) , I noticed that in Tasks the announcements are channelled through the Notifications one while in OpenTracks it's through the Media output thus I can set the volume separately for media player and announcements for Tasks but for OpenTracker the volume is changed for both media player and announcement through the media setting ( or the button on the side of the phone for that matter )

serpico, 2019-11-15:
> I actually thought this functionality was in place (but I am not sure).
> Could you check what happens if you have call in parallel?
> Because for this case, I saw the implementation (although never tested it).
> 
> Some more infos: https://stackoverflow.com/questions/11275400/reliably-pausing-media-playback-system-wide-in-android

I started by this test but I couldn't hear any announcement out of OpenTracks while while on call ( called my cell with the landline, leaving the handset near the TV speaker with the TV running and went outside walking with OT running, Voice Frequency set on 1 min but couldn't hear any announcement from OT during the 5 minutes I did the test.

After that I did the tests with the media players, same Voice Frequency and it worked
I'll retry the test on call ( after killing the app, after shutdown and restart the phone ) to confirm the behaviour is the same ( no announcement during calls )

dennisguse, 2019-11-15:
The notification channel sounds about right.
The stream is selected here: 
https://github.com/dennisguse/OpenTracks/blob/87d92d25f9c45e22777e527a9775c18d4849d470/src/main/java/de/dennisguse/opentracks/services/tasks/AnnouncementPeriodicTask.java#L196

And `TextToSpeech.Engine.DEFAULT_STREAM` is mapped to `AudioManager.STREAM_MUSIC`.
So, we could move it to `AudioManager.STREAM_NOTIFICATION` and that should make this possible.
@serpico Would you be able to test this?

About the call.
Good to know that this code works as expected.

serpico, 2019-11-15:
> @serpico Would you be able to test this?
Sure...which way is the most hassle-free for you? I mean I have Android Studio on my machine but I can't do shit with it :-) basically what I'm able to do is install/uninstall apk, provide logcat using cli and...that's about it haha

dennisguse, 2019-11-15:
You can just replace the TextToSpeech.Engine.Default_Stream with AudioManaer.Stream_Notification and compile it

closed 2019-12-04 by dennisguse