OpenTracksApp/OpenTracks #77: For Fdroid: add French translation

dennisguse, 2019-11-24:
Translation for Fdroid's description needs to be translated to French:

https://github.com/dennisguse/OpenTracks/blob/master/fastlane/metadata/android/en-US/full_description.txt

edern-appere, 2019-11-24:
Hi Dennis,
I can work on that issue but I'm a total newbie concerning how Github works. Actually I only know how to post bug reports or comment on bug reports, not handle file creation and so on. When I'm done translating, the easiest thing for me would be to post the french translation as another comment on this issue if it's ok for you.

edern-appere, 2019-11-24:
Here is the French text :

OpenTracks est une application de suivi d'activités sportives qui respecte totalement votre vie privée.

Fonctionnalités :
* Suivi d'activité : enregistrez vos activités sportives et de plein air
* Photos et repères : ajoutez des photos et des repères pour marquer des endroits intéressants pendant que vous enregistrez votre activité
* Rythme cardiaque : support des capteurs Bluetooth LE
* Export : export des traces aux formats KML ou GPX
* Pas d'accès internet requis : Internet n'est pas nécessaire au fonctionnement de l'application

edern-appere, 2019-11-24:
If you also need French translations for other parts of the app let met know.

dennisguse, 2019-11-24:
> If you also need French translations for other parts of the app let met know.

Thanks for the offer - at the moment, I am still reworking large parts of the application and don't really what is there and what will be needed.
I already created an account on transifex on which translations can be added: https://www.transifex.com/opentracks/opentracks/
But I must admit, I never used it so far :)

dennisguse, 2019-11-25:
> Here is the French text :
>
> OpenTracks est une application de suivi d'activités sportives qui respecte totalement votre vie privée.
>
> Fonctionnalités :
>
>     * Suivi d'activité : enregistrez vos [...]

Awesome.
You can just upload it via the web interface (see https://help.github.com/en/github/managing-files-in-a-repository/creating-new-files).
The following path is correct: `fastlane/metadata/android/fr-FR/full_description.txt`

edern-appere, 2019-11-28:
I've created the file. It's my first ever commit 8-)

dennisguse, 2019-11-29:
@edern-appere I saw the commit yesterday in your fork of  OpenTracks (i.e., https://github.com/edern-appere/OpenTracks).
To include your change, you need to open a pull request, so the commit is added there as well.

You can find more information here:
https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request

PS: How did you create the commit?
Via the web interface or via local git client?

edern-appere, 2019-11-29:
Ok, I was not aware I had made my own OpenTracks fork... 
I made a pull request, I hope I made it right
I'm doing everything from the website since I do not master Git at all

dennisguse, 2019-11-29:
No worries :) and thanks!
Your description will be in Fdroid with release v3.2.8 (available at Sunday I guess).

PS: If you are on Windows or MacOS, I can recommend [Github Desktop](https://desktop.github.com/).

closed 2019-11-29 by dennisguse