OpenTracksApp/OpenTracks #79: Add OpenstreetMap

pstorch, 2019-11-24:
This app had Google Maps in the past. Is it an option to add OpenstreetMap as an replacement? Could work with offline maps as well.

dennisguse, 2019-11-24:
Yes, you are right - Google Maps was included in MyTracks.

However, I am (at the moment) not a big fan of introducing a map into OpenTracks.
Besides adding technical complexity (and there is still a big amount of legacy left), it usually requires Internet access.

What functionality do you need: showing a map while recording or showing the track on a map afterwards?

For now, OpenTracks can send tracks to ask apps to show the data (KML/KMZ via ACTION_VIEW).
This works well with OsmAnd~ (shows even some diagrams) and also MapsME (sadly, also imports the track).
Only issue with OsmAnd~ is (at the moment) that the loading process cannot be aborted (see https://github.com/osmandapp/Osmand/issues/7967) and it is rather slow.
This also works while recording, although the track is not updated in the app that is showing it.

This functionality is available via the menu (small map icon):

[image]

Anyhow, if you have time and know an awesome solution, I am happy to integrate it.

pstorch, 2019-11-25:
My main usecase is to look at the track on the map afterwards. Up to know I opened the track with Locus Map and it always wanted to import the track in some of its folders. That was a bit cumbersome. With OsmAnd it's easier and faster. Thanks for this suggestion.

I understand that this feature is not necessarily one of your goals and it adds complexity. I would treat it as a nice to have comfort feature.

For another app I introduced openstreetmap via mapsforge API. There you can choose between online maps (which requires mobile data) or offline maps (which requires some configuration steps to be done by the user).

dennisguse, 2019-11-26:
I just slept a night over your request.
What do you think about adding this as a plugin (like https://github.com/siacs/ShareLocationPlugin).
This has the advantage that the main application does not need `android.permission.INTERNET`.
This way every user can decide what is actually required...

I don't know if this is technically possible, but it would be great to (re-)add this as a tab in the `TrackDetailActivity`.
However, I have no clue (at the moment) how to actually implement this.

pstorch, 2019-11-26:
I've seen App Features extracted as separate plugins, but never developed one myself. 
Maybe I find some time to look into it. That looks like an interesting approach.

pstorch, 2019-11-30:
Up to know I could only find out how to check if a PlugInApp is installed (via PackageManager) and call one of its Activities via Intents. But I couldn't find a way to use Fragments with their own layouts from another App. I doubt that this is possible.

dennisguse, 2019-11-30:
Calling an activity would be possible - it could just show the map as well as the statistics dashboard. One practical issue would be the connectivity to the TrackRecordingService / Database to get live updates about the state of the track (while recording).

pstorch, 2019-12-01:
> One practical issue would be the connectivity to the TrackRecordingService / Database to get live updates about the state of the track (while recording).

I did a small proof of concept and started another activity with the content URI of the trackpoints table. The CustomContentProvider already offers this functionality. I was able to receive updates while the TrackRecordingService was running.

dennisguse, 2019-12-01:
Nice!
One option would be to provide the functionality of the TrackDetailActivity  as a plugin. Then the user could select which one to use (i.e., default vs one with a map). 
That would also allow to provide different layouts of the stats page / charts etc.

pstorch, 2019-12-01:
That could also be a solution. But then I guess common functionality   should be extracted into a library project. Otherwise there will be lots of  duplicate code.

dennisguse, 2019-12-02:
That's for sure!

@pstorch Did you have a specific architecture in mind?
And did you need to export the TrackRecordingService or something else?
And could you share your code?

pstorch, 2019-12-02:
> @pstorch Did you have a specific architecture in mind?

No, I haven't worked with library projects yet.

> And did you need to export the TrackRecordingService or something else?
> And could you share your code?

The PoC is already shared:

- https://github.com/pstorch/OpenTracksOSMPlugIn
- https://github.com/pstorch/OpenTracks/tree/79_osm_plugin I hijacked the "Show on Map" function for the time being

dennisguse, 2019-12-04:
@pstorch I just looked into the code and it looks good.
A very cool solution to use a `ContentObserver` - that is quite beautiful.

So going with a separate activity works!
Only one thing would need to be improved: hardcoding the activity's name would be rather difficult.
Could we move to implicit intent (e.g., defining a custom action like `ACTION_OPENTRACKS_SHOW_MAP` [just made up])?

maxolasersquad, 2019-12-05:
FWIW, the current map button does a great job of displaying the workout in OsmAnd.

[image]

pstorch, 2019-12-06:
> Only one thing would need to be improved: hardcoding the activity's name would be rather difficult.
> Could we move to implicit intent (e.g., defining a custom action like `ACTION_OPENTRACKS_SHOW_MAP` [just made up])?

The hard coded package name was only used to hijack the map button. The actual connection is done via the mimeType `vnd.android.cursor.dir/vnd.de.dennisguse.trackpoint` when starting the intent. 

But I found another way: we can actually prepare the Action View Intent and check if an app is installed which can handle the intent. Of course the calling of this activity can be placed somewhere else.

pstorch, 2019-12-07:
There is a first version of a map implementation which shows the track.

dennisguse, 2020-01-02:
Functionality was integrated into OpenTracks.
A first draft for the map dashboard is also worked upon: https://github.com/OpenTracksApp/OSMDashboard

closed 2020-01-02 by dennisguse