OpenTracksApp/OpenTracks #82: Heart Rate recorded but not shown

serpico, 2019-11-28:
**Describe the bug**
Description looks similar to #50 
While recording the heart rate is not shown on the Statistic tab see :
[image]

Although when displaying the graphical tab it seems to be recorded :+1: 
[image]

**Technical information**
 - Device: Samsung SM-G361F
 - OS: Android 5.1.1
 - Version v3.2.6

dennisguse, 2019-11-28:
Thanks for the report.
It would be very helpful, if you could figure out how to reproduce this problem.
For example, did you put OpenTracks in the background (in between using another app) or did navigate within OpenTracks?

pes4murai, 2019-12-02:
[image]

Same here, using OpenTracks not in background but leaving the system always on during the run with the app opened

dennisguse, 2019-12-02:
Is the heart rate not shown at all or does it vanish?

PS: No changes in v3.2.8 that would affect this behavior (should be on Fdroid tomorrow).

dennisguse, 2019-12-03:
Sometimes coding in the evening is not a good idea.
While cleanup, I removed the code to set the heart rate sensor value to the UI...
Fix will be included in v3.2.9.

closed 2019-12-03 by dennisguse

bschwehn, 2023-01-24:
I have the same issue in v4.2.0 / 5119. Heart rate is recorded and shown in the graph, but the heart rate widget in the dashboard/stats view never displays the BPM.

dennisguse, 2023-01-24:
@bschwehn actually no; see #1426. 
In V4.2.0, this functionality was removed as some refactoring was required and needs to be reimplemented. 

You could use remain using v4.1.0 for now until this feature is re-added.

bschwehn, 2023-01-24:
@dennisguse Ah, great! I downgraded to 4.1.1 for now and the heartrate is showing now.

Thanks a lot for all the work you've put into opentracks!