OpenTracksApp/OpenTracks #87: Show pace

sedrubal, 2019-11-30:
**Is your feature request related to a problem? Please describe.**
A common size for runners is the pace (min / km).

**Describe the solution you'd like**
OpenTrack should display it additional to or instead of the speed. You could maybe toggle between speed an pace by tapping on the speed section.

dennisguse, 2019-12-03:
Would you like to see only the pace or also the avg/min/max?

About toggling: this is usually a functionality that is hard to discover for new users.

dennisguse, 2019-12-03:
And this is blocked by #15.

sedrubal, 2019-12-03:
I think it would be interesting to see the pace in avg/min/max or even a list of paces for each kilometer?

And you are right that toggling might not be intuitive.

dennisguse, 2020-02-14:
What do you think about not using either pace or speed but rather pace (yes/no) and speed (yes/no)?

dennisguse, 2020-08-07:
@rgmf Did we not do something about this?

rgmf, 2020-08-07:
@dennisguse no, but for runners it's an important feature (I miss it when going out for a run). In TrackRecordedActivity we should add the list of km with its pace. I'm going to look into it...

rgmf, 2020-08-08:
For run pace is showed several versions ago.

> I think it would be interesting to see the pace in avg/min/max or even a **list of paces for each kilometer**?

I'm going to check it out the possibility of show a list of pace for each km/mi because it's a feature I miss when I'm going out for a run.

dennisguse, 2020-08-08:
@rgmf I removed this functionality due to technical reasons: statistics were stored as special waypoints. It was quite a while ago (can't search the commit easily on a phone sorry), but the implementation in TrackRecordingService was cool. I just did not like the complexity of storing data. The issue to track this is #134.

rgmf, 2020-08-08:
@dennisguse my idea is not store anything but show a ListView in TrackRecordedActivity with pace by km, 2km, 3km... or whatever the user select. The pace stats by km can be extracted from TrackPoints.

Anyway I'll read #134 at home.

commit fc405c05 (2020-08-27): Interval stats: while user is recording a track intervals are refreshed every 5 seconds.
Interval stats: completed the list of split intervals: 1, 2, 3, 4, 5, 10, 20 or 50 km or mi.
Fixes #87.
commit d56f652d (2020-09-02): Interval stats: while user is recording a track intervals are refreshed every 5 seconds.
Interval stats: completed the list of split intervals: 1, 2, 3, 4, 5, 10, 20 or 50 km or mi.
Fixes #87.
closed 2020-09-02 by rgmf