OpenTracksApp/OpenTracks #90: Choose save location

maxolasersquad, 2019-12-04:
**Describe the solution you'd like**
I would like to be able to choose a location where the tracks are saved. Ultimately I'd like to use the file chooser to save to my Nextcloud server. I do this with KeePass to keep my password file backed up and synced. The intention here is that my tracks would be backed up and synced across devices. This solution should allow automatic cloud backups without the app itself requiring Internet permissions or cloud support directly. 

**Describe alternatives you've considered**
Direct integration with Next/Own cloud would be easier, but would violate the privacy intentions of this app as I understand it.

dennisguse, 2019-12-04:
Nice idea. 
Is your goal to sync multiple devices or just backup
 the data?
What do you think about the following solution:
1.  After finishing a track (i.e., pressing the stop button), the track is exported automatically to the external storage (I.e. /OpenTracks/FORMAT/trackData)
2. Next/OwnCloud can monitor this folder and then upload the files

Then you could also automatically download to another device and import it automatically to OpenTracks. 

However, this is not implemented right now. 
One assumption in OoenTracks is that after finishing a track it won't change. 
For now this is (almost) true, but might change in the future. 
The following issues would need to be solved:
1. export must be triggered manually (in settings)
2. import must be triggered manually
3. configuration option for the user (mainly if and which file format)
4. OpenTracks does at the moment not overwrite files that were already exported (it adds a number to the filename and exports it again)

All of this is doable and I don't see any issues with integrating it this way. 

Direct integration with any Next/OwnCloud would be out of scope (you are right).
Or if you really want it, it could be implemented as a plugin.

maxolasersquad, 2019-12-05:
Both sync and backup are my goal. I'm currently using Run Keeper and have years of workout data stored with them. I'd like to self-host my workout data with a FOSS app like this one and keep it archived.

Ideally I'd like to also create a Nextcloud app that could display the workout data.

dennisguse, 2019-12-29:
Finished migration to Storage Access Framework: import/export functionality.
Does not allow

MrS1R1US, 2020-04-14:
>     1. After finishing a track (i.e., pressing the stop button), the track is exported automatically to the external storage (I.e. /OpenTracks/FORMAT/trackData)

I don't have link "/OpenTracks/FORMAT/trackData", only Android/data/de.dennisguse.opentracks/files/Pictures/(clear). Where are training files stored to automatically duplicate them on my computer? Clicking "export" in the settings is a big time consuming.

dennisguse, 2020-04-14:
No work on this issue was conducted so far.
@MrS1R1US Only the pictures are stored on the file system (app-private).

jancborchardt, 2020-06-19:
Hi from [Nextcloud Maps](https://github.com/nextcloud/maps/) in the name of @eneiluj @tacruc and me! :)

We’d love to collaborate on some closer integration with OpenTracks. Is sync integration still seen as out of scope, and only workable as a plugin? If we can help somehow let us know, @eneiluj @tacruc will have more insight regarding APIs and such on our side.

dennisguse, 2020-06-19:
Sounds awesome.
One thing is that Internet access from within OpenTracks is (at least from my perspective) out of scope. So, some other app has to take care of uploading and downloading while OpenTracks takes care of (automated) import/export (other apps might be NextCloud/Syncthing etc.).

What I can imagine is that NextCloud synchronizes an _OpenTracks folder_ on external storage while OpenTracks (automatically) does import / export to this folder.
Manually, this does not work as the import functionality does always _add_ the tracks while it would need to overwrite/ignore already existing tracks and not import those as new tracks.
What we lack here is a global identifier for tracks.
IMHO, we need to do three things (for real synchronization)
1. somehow generate a globally unique identifier (that does not allow to identify/track an OpenTracks installation; so no prefix). Something like a UUID?
(Such a thing would not be rocket science, it is just tedious work as the trackId is at the moment defined as long and used everywhere in code base incl. tests). See also #276.
2. Change import to actually ignore/overwrite already existing settings (we might need some user feedback for this, if a conflict occurs)
3. And later on we need to automate this functionality, but that would be a second step.

How does this sound?
So, an MVP style would be to implement 1+2 and check if it works as desired (manually) and then figure out how to the export automated.

PS: What I would actually love would be synchronization via Bluetooth as my devices usually are close by. But I guess the user value for this is often rather limited ;)

@jancborchardt @eneiluj @tacruc Awesome!

jancborchardt, 2020-06-19:
cc @tobiasKaminsky @AndyScherzinger from our Android app who will know more about how or if the approach you describe would work with our existing app. :)

AndyScherzinger, 2020-06-19:
The Nextcloud Android client doesn't have a real sync while its auto upload feature offers custom folders which would detect changes and upload the files to the Nextcloud server, it doesn't support deletions though.

If that is enough for this use case then that is already available.

dennisguse, 2020-06-20:
So, we have the following use cases:

* Automatic backup (as requested in #186): export tracks to a internal storage and let the NextCloud Android app handle the upload (@AndyScherzinger thanks for the input).

* Synchronization: in addition to backup, we need automatic download from the backup destination (via 3rd-party app).
Handling deletion might be a different story (but probably required somehow).
One alternative option here would also [SyncThing](https://f-droid.org/packages/com.nutomic.syncthingandroid/)

One issue for synchronization would be conflict resolution.
And that is a tough one - I hope this will be handled outside of OpenTracks.
If not, we have a lot to learn :)

@AndyScherzinger Are there plans to implement something like folder synchronization into the NextCloud Android?

And I created a project to track backup/sync-related issue/tasks: https://github.com/OpenTracksApp/OpenTracks/projects/3

AndyScherzinger, 2020-06-20:
There is a simple, configurable conflict resolution configurable for a folder for Nextcloud's auto upload.

As for the folder synchronization, yes, that is a planned feature while due to limited Dev capacity might not be implemented in the near future. Maybe @tobiasKaminsky can shed some light on this one.

dennisguse, 2020-08-15:
Recently, we added a UUID to each track and, as of #312, these will be exported/imported.
If a track (by UUID) is already existing, then (by default) the import process is aborted.
The UI needs to be improved, but at least, we can now "manually" synchronize multiple installations of OpenTracks (also on the same device *yeah*).

jdohm, 2020-11-29:
@dennisguse maybe a bit much to ask for, but is there a option to prevent OpenTracks from re-importing existing tracks that where recorded outside of OpenTracks?

For example: I use the Nextcloud Backup solution. In the last Days where the OpenTracks App had some crashes I exported the Tracks from PhoneTrack (a fulltime logger for Nextcloud) into the sync folder and clicked "import all". 
As I did this the second time, OpenTracks imported the newly added Track, but also re-imported the old one. 

As this is probably a niche example, I'm fine with just deleting the exported old Track, but wanted to let you know, just in case there is an easy fix you can thing off.

dennisguse, 2020-11-29:
@jdohm we generate the a unique identifier per track (UUID) and on import check if this is already in the database. If there is none, we generate a new one. 
Just import the track once and export it (manually replacing the file).

jdohm, 2020-11-29:
@dennisguse thanks for the quick replay and solution.

As this is somewhat complicated (even if the user knows he has to export and override the imported file to prevent re importing, he most likely has to delete the imported file and rename the newly exported one). 

Maybe a yes/no question if the user want's to add the UUID to the imported file to prevent re importing, after importing a track without a UUID, would be a nice addition. 
This is probably some work and as I said a niche case, but wanted to let you know in case there is a time in which there isn't more important work to do.

Thanks for your work on this already great app :) always nice to see such active people on a open source project.

pstorch, 2020-11-29:
@jdohm this only happens for exports from version prior to v3.9.0, so I don't know if it is worth the effort to write the created UUID back while importing. This would mix Importer and Exporter logic.

@dennisguse would it be worth to start an FAQ? I don't know if this will be a "frequent" question though.

dennisguse, 2020-11-30:
@jdohm Can you create a new issue for this feature (incl. the info discussed here)?

rkozlo, 2023-08-11:
Hi, seems similar so won't create new issue. I set automatic export to my nextcloud server after finished activity. Sometimes it stops working "Can't be saved"(or something similar). Manual export tracks working. Choosing the same destination directory and next activity may be send or not. Seems not to be determinstic. Is it known issue maybe? Or could i turn on some debug mode to gather some logs maybe?

maxolasersquad, 2023-08-11:
This feature now "works" because in the export directory I am now able to use Nextcloud as a destination. I believe this is a newer feature of Android since I posted this, or this app is using a different Android API to that has such an allowance.

dennisguse, 2023-08-13:
@maxolasersquad sounds great. I wasn't aware of this actually and just checked it! Really nice :D

@rkozloI could you open a new issue for your problem?
https://github.com/OpenTracksApp/OpenTracks/issues/90#issuecomment-1675253562

closed 2023-08-13 by dennisguse