OpenTracksApp/OpenTracks #97: Promo codes for Google Play Store

dennisguse, 2019-12-10:
Promo codes to install OpenTracks from the [Google Play store](https://play.google.com/store/apps/details?id=de.dennisguse.opentracks.playstore).
Instructions on how to redem can be found [here](https://support.google.com/googleplay/answer/3422659).

Please mark the ones that were used (valid until End of December 2019):

- [ ] 8SG6HTKZSED1A4BS1NKFDC6
- [ ] 9SW63KRA8UQUS1836M6D7DE
- [ ] 13PSK97EJY3UN9U1CR7ES2D
- [ ] XUK8APQDF84F0MLAC71N590
- [ ] K7D0PHDZTSKAMC517LFW27M
- [ ] R544JZXDKDATJ4X5AF31HVG
- [ ] PUHWW6KM5T80T4GSDDZT9G5
- [ ] ELKYL6CMPQXMKXSUWDLXLMJ
- [ ] V6L7L0S9TGJ4LD70XJSJ6EG
- [ ] 03BY5WMS8GV3KWGA1T75CW2
- [ ] T1JZCJTSGRZSHTGQR6SFW60
- [ ] 2F3ZDKE3N2444LSN3WQEJBX
- [ ] WNGQC4B682DLNN1CZYAUTUD
- [ ] H610CB3GSZL6SLCVZ1M0B7X
- [ ] MVNXKSUQ312963DSF69R3HT
- [ ] 7YM48FUJ79SYQH13DFA9UFV
- [ ] ZSCDA8J8NE7SGXFY993ME3F
- [ ] 8RLASXRQCKMMEM5M4T6QJJQ
- [ ] MKTZQS0FCZ3YBEQMDYHGR74
- [ ] 9KCUZ76F6RGWE253AJ73TR2
- [ ] KDGQ1G0WH5N516E2644BZK2
- [ ] PCJVEQEBJ2AW3J1UBNRB6PH
- [ ] 2RCRV6HXNTM4DXDWMDZ8QSR
- [ ] 9J9K8GXWF79J3C01TE4E4PQ
- [ ] FGX5STY5PJ621Q4UQKFBTS0
- [ ] 3ZDK4UF4K4SGU8FX6V19ZCX
- [ ] WU3QD6JXDRTRQ33UB3CAUGE
- [ ] LQ7Y222MBMPVXKP2C8CX36K
- [ ] 851KAHTVZP6L2RNN1CPR47R
- [ ] 7D2F666FFEHZY6NW0BCNRYK
- [ ] YM5R5014KYUCDMGQGH5ECR4
- [ ] HUBXDR3W3MWEQ0QU0CC9YVV
- [ ] UHUD0TNACT58VG2CEZTVCL2
- [ ] D01E3FDB4A8UZZGFL7R25SH
- [ ] WMB570LTG31NR8V9TUDPU5R
- [ ] 24FRWDR38UKWTYGZ8XGRGP7
- [ ] GUK96PB59JQDH33UN1VYGNS
- [ ] 561LFF2YX0PWPT9Y8J8B1A6
- [ ] Z7WY0RSN2W88A8VSL1YK779
- [ ] 2NZU96HBCJ4LSTDUGUMC92F
- [ ] S70MTEV0BWVCQGCZ7WZ3LUW
- [ ] 4PB65AKX36DJCQRQEGJR0HQ
- [ ] 0RU8A9NN2VWS8HRWZYD3GG6

pes4murai, 2019-12-17:
Dear Dennis,
thank you for the Promo Code.
I've used 4PB65AKX36DJCQRQEGJR0HQ but I can't mark it in your list.
Thank you so much!

noodle-, 2019-12-30:
I used this one: WNGQC4B682DLNN1CZYAUTUD
Thank you!

audunmg, 2019-12-30:
Thank you! I used this one: 13PSK97EJY3UN9U1CR7ES2D

Dabombber, 2020-04-11:
Used 209GRG9RA6R7M7TCKG2CCA5, thanks.

GajenderY, 2020-04-16:
Thank you, I used HQ9GY71FD14MUJXCYT0XC4P but wasn't able to mark it.

blackris-github, 2020-06-04:
Thanks, tried the f-droid version but it couldn't get a GPS signal. I will play around with the settings a bit and try out the Play Store version. Used the code:

WVWTVLQGAW2A7QDRSS9RAUF

dennisguse, 2020-06-04:
@blackris-github The PlayStore release is identical to the F-Droid release.
If you have any problems just open a new issue - and we can try some troubleshooting.

xax, 2020-06-15:
Used `HZUPAZLKGGFH32TQSQBTQER`; found `66WLM97429YYT1P8UHVVNH4`
 having been used previously.

Thanks a lot.

SHMAUS-Carter, 2020-07-14:
Used 4GPSDEYNZGW67LMDHYB318M

lannetscan, 2020-08-20:
Used MQB58N3UBY6TW68UEK2TC2K
I really enjoy your app, but the fdroid package could not be parsed (android 11)

sttoma, 2020-09-01:
Thank you, I used 
2Q1E6CDJGVGRWFP2NUZ6EEJ
but I can't mark it in your list.

MarcoDjallo, 2020-09-12:
4ELML5L5NRHE9XP0APK325L used! Thank you

nemeises, 2020-11-22:
hello dear dennis please give me gift card google play i very very need to this
please 🙏
tank you

nemeises, 2020-11-22:
nemesises94@gmail.com

frqusg, 2020-12-25:
please give me gift card google play i very very need to this
please 🙏💔

bedri87, 2021-01-06:
please give me gift card google play

Ra242, 2021-01-07:
Hey bro please give me unused promo code

kiraci03, 2021-01-08:
Can you give me one that work of them please? I went through a lot and all (or nearly i guess) are used

jackprink, 2021-01-15:
I have list of code that work and real. but you need to spend your time and complete some basic task so you earn points and get your google play gift card reward.
follow below link to get it now -[https://www.freereward.net/google-play/](https://www.freereward.net/google-play/)

closed 2021-03-01 by dennisguse