OpenTracksApp/OpenTracks #103: TrackDetailActivity should start an actual recording

dennisguse, 2019-12-15:
Right now, `TrackListActivity` starts the recording (i.e., trigger `TrackRecordingService`).
This leads to some weird animation (especially on slower phones updating the list of tracks (and resorting it).

It would be better to move this functionality to `TrackDetailActivity` as this activity is anyhow showing the actual data.

dennisguse, 2020-02-27:
Actually, it would also be a good idea to split TrackDetailActivity into one that is doing the actual recording (also starting the recording) and one that just shows the details afterwards.

rgmf, 2020-03-21:
> Actually, it would also be a good idea to split TrackDetailActivity into one that is doing the actual recording (also starting the recording) and one that just shows the details afterwards.

@dennisguse  I'm checking this issue out like you know and I think split TrackDetailActivity in two activities is really important to fix this issue. Otherwise the logic and maintenance become more difficult.

So, I'd split it in: TrackRecordActivity (for showing while recording) and TrackDetailActivity (for showing finished tracks). Also each one with own fragments: StatsRecordFragment/ChartRecordFragment and StatsDetailFragment/ChartDetailFragment.

After doing this the issue becomes easier one (altough isn't not simple because there are a lot of things implied).

So @dennisguse, What do you think?

dennisguse, 2020-03-21:
@rgmf Sounds pretty awesome and also like some pretty big change.
I am not entirely sure, if need to duplicate the fragments is immediately necessary.
Especially, the chart fragment would not provide different functionality, right?

One issue will be to manage the activity stack (especially what does the back button do).
After stopping a recording the TrackRecordActivity must be replaced by a TrackDetailActivity and the other way around when continuing a track.
Some more information is available here: https://developer.android.com/guide/components/activities/tasks-and-back-stack
But I have not used this functionality.

rgmf, 2020-03-21:
> @rgmf Sounds pretty awesome and also like some pretty big change.
> I am not entirely sure, if need to duplicate the fragments is immediately necessary.
> Especially, **the chart fragment would not provide different functionality, right?**

Yes, I think so. But, in fragments TrackDetailActivity is used for getting the track data hub:
`trackDataHub = ((TrackDetailActivity) getActivity()).getTrackDataHub();`

For now I can use an Interface to share Stats and Chart fragments.

> One issue will be to manage the activity stack (especially what does the back button do).
> After stopping a recording the TrackRecordActivity must be replaced by a TrackDetailActivity and the other [...]

Thank you for the link. It'll be useful for me.

This drawn describe what I thougth to do and I think is the same you point out, isn't it?

[image]

dennisguse, 2020-03-21:
Looks good to me - the only thing after "finish click" (leading to DetailActivity) should the back button lead to ListActivity.

And internal data passing is not yet as cool as it could be.
TrackDataHub reads gets the data directly from the database (and is thus decoupled from TrackRecordingService) and mainly provides resampling of TrackPoints.
I am not really happy with it as (in my head) it makes life rather complicated.
But is is used by stats and charts...

rgmf, 2020-03-23:
@dennisguse I need to make a decision.

So far now, I have splitted the TrackDetail in TrackDetail and TrackRecord. The only thing left is move service creation from TrackList to TrackRecord.

I can backward, but what I have done is split StatsFragment too. Why? Because:
- TrackDetail not need implement `ChooseActivityTypeDialogFragment.ChooseActivityTypeCaller` (the activity type is changed in TrackEdit).
- I think TrackDetail doesn't need neither `TrackDataHub` nor `TrackDataListener`. I think is simpler that the fragment get the trackId in `TrackDetailActivity` and loads the layout with `TripStatistics` created with the `Track` object directly without `TrackDataHub`.

In other words, having TrackDetail and TrackRecord sharing the StatsFragment complicates TrackDetail with things it don't need.

Another option can be to add to TrackDetail the option of change the activity type directly (like in TrackRecord) and using TrackDataHub in TrackDetail too.

Two questions:
1. do you think new `TrackDetailActivity` must use `TrackDataHub` to load `Track` data in the fragment? 
2. What do you think about it? You know the code better than me and I don't know if I misunderstanding something.

dennisguse, 2020-03-23:
@rgmf That are some really good questions.
Without looking into the code ChooseActivityTypeCaller should not be necessary.
But if I recall correctly the TrackDataHub is required for the ChartFragments (to downsample the number of points).

So for now I would keep TrackDataHub there (but there is some big room for improvement).
I am also not a big fan of the TrackDataListener as for my feeling the interface is too big (leading to empty implementation).

PS: I did no yet touch this code in detail, so I guess you already are deeper in it than I am :)

rgmf, 2020-03-26:
@dennisguse, finally I decided to create an Activity between List and Detail. I think is the best and more elegant option. I've been studying the code around recording (Services, TrackDataHub, Listeners...) and the more I deep in it the more I like it.

Activity between List and Detail starts, resumes, and finishes track recording. Also, this new Activity starts Service.

 Let me test it deeply and I'll pull request.

With these changes #103 and #11 would be fixed.

commit d93c2ec5 (2020-03-30): TrackDetailActivity starts an actual recording and a stopped recording can be resumed. Fixes #103 and #11.
closed 2020-03-30 by dennisguse