OpenTracksApp/OpenTracks #106: All pace times same value

ml-, 2019-12-18:
**Describe the bug**
When finishing a track the final pace times displayed are all the same (1st screenshot).
When switching to speed (km/h), the effect is similar (2nd screenshot).
Shortly after v3.2.10 was released I wanted to go for a run and check out the new version, it wasn't on F-Droid however, so I compiled it from the tagged commit (e77590a125222c515cb9ff699a3081796d237116) and installed the debug apk since I didn't want to deal with apk signing.

After doing some quick math, it seems very likely that the displayed value is the actual average.

Can ensure that I didn't move at the same speed for 14.5km (which would be astonishing tbh). I can remember the voice telling me it was way lower in the first few km.

I know for sure that the issue was not present in v3.2.9. Since I only tested my MAY_DUCK change without moving, I also can't say if it is before or after it.

Maybe you have an idea where this might have slipped in.

**To Reproduce**
Didn't go a second time yet, so can't tell if this is reproducible. 

**Technical information**
 - Device: Nexus 5X
 - OS: LineageOS 15.1
 - Version v3.2.10 (debug)

[image]
[image]

dennisguse, 2019-12-19:
i can reproduce the issue.
As this release was not yet discovered by F-Droid: I just remove the release tag for v3.2.10.

Quick check into the code: it does not seem to be an issue of `StatsFragment` that sets the value.

dennisguse, 2019-12-19:
@ml- Can you check what information you get, when you export these tracks?
It should be the identical:
* if it is then it is a computation issue within TrackRecordingService(?)
* if it is not it is a UI issue.

This issue might be related to the removal of the `WaypointType.STATISTICS`.
These waypoints could be automatically added to store the statistics until a point of a track (on time interval or distance).

At the moment, I cannot rely reproduce the issue.
I thought I could but did not rely reliable.
In addition, I saw some weird behavior that the maximum speed actually decreased...

dennisguse, 2019-12-19:
Additional insights:
* The data should be assigned to a Track
* On loading a track from DB (via ContentProviderUtils), Track.tripstatistics is initialized with DB data
  (same table)
* DataSourceManager (TrackTableObserver) notifies TrackDataHub when a change in the track table occurs
* TrackDataHub notifies StatsFragment when a track is updated  (written to DB?)
  and loads it new from the DB before passing it to StatsFragment.
* StatsFragment uses this object to update view

But I still have no idea what is going on :(
The code has a "character"...

ml-, 2019-12-19:
It's only one track and I have exported it. It seems fine looking at the data. Only one thing seems to bug the application GPXSee where it simply complaint about the invalid coordinates. Other applications like gpx-viewer or OSMAnd display this fine.

In gpx:
```xml
<wpt lat="100" lon="180">
<time>1970-01-01T00:00:00.000Z</time>
<name><![CDATA[Marker 0]]></name>
<cmt><![CDATA[STATISTICS]]></cmt>
<desc><![CDATA[]]></desc>
<type><![CDATA[]]></type>
</wpt>
```

In kml
```xml
<Placemark>
<name><![CDATA[Marker 0]]></name>
<description><![CDATA[]]></description>
<TimeStamp><when>1970-01-01T00:00:00.000Z</when></TimeStamp>
<styleUrl>#statistics</styleUrl>
<Point>
<coordinates>180.0,100.0</coordinates>
</Point>
</Placemark>

dennisguse, 2019-12-19:
Nicely spotted.
That waypoint should actually not be exported and is a actually the result of a `WaypointCreationRequest` to create the `TripStatistics`.
https://github.com/dennisguse/OpenTracks/blob/20f1524701a3ae26b74ec84e3eb822b2b6ca701b/src/main/java/de/dennisguse/opentracks/services/TrackRecordingService.java#L330

I have work in progress that removes/rewrites the functionality, but I need to do it again as it just one big chunk of rework (basically some lessons learned).
Basically, I want to remove this "special" waypoint as it is not really a waypoint (in terms of a marker created by the user).
It is actually used to store the summary of a track - which actually should be stored with the track itself (which it is if I understood if).
The main issue is that this also touches the import/export code and the test coverage for this is 0%.
So, it would be great to add tests first (and fix #104).

dennisguse, 2019-12-25:
I added some tests and removed the statistics waypoint completely.
However, this does not fix this issue.
I could reproduce it only on shorter tests: max speed and avg moving speed are always identical (while the later is presented for both).
However, it does not occur always: my guess is that I introduced by a side while cleaning up (likely in `TrackDataHub` as `StatsFragment` presents this data; not from GPS directly.

@ml- 
Avg speed is not afffected; you just made no break ;)

ml-, 2020-01-06:
Sorry for the late response, been to the congress where I surprisingly found some OpenTracks stickers. Didn't meet you however. Maybe next time.

You're right, "avg. pace" and "avg. moving pace" are fine. I started avoiding traffic lights recently so that there is no longer a need for a break, that's what confused me I guess.

I could not reproduce this with v3.3.1 yesterday, same route same distance. Although something else bugged me (voice announced pace lower than final fastest pace), I don't think it's part of this bug. Might as well be mistaken. Will try this again at the end of the week I guess.

dennisguse, 2020-01-06:
Thanks for taking the time to test it.
Just close the issue if you can not reproduce anymore :)

About Congress: cool that you saw the stickers (marketing worked ><).
Next time, just drop me a note ;)
It was too short as usual...

closed 2020-01-06 by ml-