OpenTracksApp/OpenTracks #112: Analyze all tracks

mwip, 2019-12-26:
This feature was removed a couple of versions ago. However, I think that it would be very useful to get a quick overview of the training effort. 

Also, it would be nice to be able to group by year, month or week. 

Do you think this would be a valuable feature?

dennisguse, 2019-12-26:
For sure - I would love to see this implementation properly!
I removed the available implementation as it was just summing up all recorded tracks: there was no differentiation by activity type, filtering or anything else.

mwip, 2019-12-27:
Awesome. I'd love to help, but unfortunately I am not much of a java developer myself. So if there is anything I could do besides coding, e.g. testing or reviewing, I'd be happy to do this.

dennisguse, 2019-12-27:
Coding is actually not that complicated in this case.
It is basically about accessing the database (via SQL) and hand the data to the UI (which is actually the complicated part).
What would be great if you could define the goals that you want to achieve (why do want statistics?) as well as a draft of a possible UI design.

Then somebody can just take and implement it.
One technical requirement: try to avoid introducing additional libraries for visualization.

dennisguse, 2020-01-02:
An API for external plugins is now available: https://github.com/OpenTracksApp/OpenTracks/commit/a3eb1070686fe3a377ab50c4a9bb1c91bd62c279

commit c1de9967 (2020-06-08): Aggregated statistics by activity type. Fixes #112.
commit 293da3d5 (2020-06-09): Aggregated statistics by activity type. Fixes #112.
commit 1cbab682 (2020-06-09): Aggregated statistics by activity type. Fixes #112.
commit 91f29599 (2020-06-09): Aggregated statistics by activity type. Fixes #112.
closed 2020-06-14 by rgmf