OpenTracksApp/OpenTracks #113: Music killed voice frequency announces statistics

slackline, 2019-12-27:
**Describe the bug**
I have Voice frequency set to 1km and when out running/cycling when the distance/time/pace are announced it stops and kills any music that is being played.  This didn't used to happen, instead the audio was over the music that is playing using LongPlayer (which is itself no longer available in the Play Store).

**To Reproduce**
1. Set Voice Frequency to anything but None
2. Start recording a track
3. Play some music
4. Wait for lap/pace to be reported, audio stops.

If applicable:
* add screenshots
* add logcat output

**Technical information**
 - Device: Samsung S7+ / Motorolg G7 Plus
 - OS: LineageOS 14 / LineageOS 16 (respectively)
 - Version v3.2.9/v3.2.9-bafc7d4-

I'll try and capture a logcat later.

dennisguse, 2019-12-27:
Thanks for the report.
I introduced this behavior accidentally in v3.2.9.
It was fixed in v3.2.10.
Could you upgrade and test if this fixes the issue for you?

slackline, 2019-12-27:
Thanks for the quick response and for having already addressed the issue by the sounds of it.

Just updated to v3.2.10. Will try it in a few hours when I head home and report back.

slackline, 2019-12-27:
I can confirm this issue does not occur with v3.2.10

My apologies for not updating and checking first.

dennisguse, 2019-12-27:
Don't worry - thanks for taking the time to report the issue.
And have fun with OpenTracks!

closed 2019-12-27 by dennisguse