OpenTracksApp/OpenTracks #115: Need new icon for markers

dennisguse, 2020-01-04:
The summary icon for markers is the KML blue push pin (`@drawable/ic_marker_blue_pushpin`).
However, this icon only available as a _low resolution PNG_ and the colors do not really fit the UI.

So, we need a replaced (best case SVG).

This is how it looks like currently:
[image]

dennisguse, 2020-05-31:
We can replace it with the OpenTracks logo.

DeusFigendi, 2020-07-14:
Here I vectorized it (manually). Colors could be changed easy.
License is cc0

[(sadly I cannot upload SVG in this issue nor .gz or .zip so I used gist for it).](https://gist.github.com/DeusFigendi/ace7c40e438be9dff262168020ba6a3a)

[image]

dennisguse, 2020-07-14:
@DeusFigendi Nice!
It would be great if you could create a pull request and put the SVG into this folder:
https://github.com/OpenTracksApp/OpenTracks/tree/main/drawable-svg/

If you any suggestions for a different coloring - please feel free to make suggestions.

DeusFigendi, 2020-07-14:
> If you any suggestions for a different coloring - please feel free to make suggestions.

Dude, no idea, I just mentioned that recolouring is easy, because you wrote it wouldn't fit the UI.
To be honest I don't even know this project, so I don't know which color scheme would fit the needs XD.

>  It would be great if you could create a pull request and put the SVG into this folder:
sure I can do so.

dennisguse, 2020-07-14:
@DeusFigendi I have no clue about colors - I always hope that somebody else has ><
The current color is black (or a really dark gray) + orange.
Thx

dennisguse, 2020-07-16:
We now have a working SVG in #320 that was already merged.
Now, we "just" need:
* to replace the blue push pin
* check that the color fits

commit 5f637952 (2020-07-18): Replaced blue pushpin icon (png) with with orange icon (svg).
Fixes #115.
closed 2020-07-18 by dennisguse