OpenTracksApp/OpenTracks #117: Stop Recording Automatically After Idle Period

nthh, 2020-01-06:
This feature could exist and I have overlooked it, but it would be nice for the app to automatically stop tracking after some period of idle time. I often find myself forgetting to stop the recording and it records too long of a period.

Example: 
After n minutes of inactivity (device not moving more than n distance), stop recording current track

dennisguse, 2020-01-06:
Thanks for the request - this functionality is not implemented (you are right).
One issues for this functionality would be that a stopped track cannot be continued (see #11).

Nevertheless, setting the track on pause would be possible.
What do you think?

nthh, 2020-01-06:
That seems acceptable, as pausing probably fits in better here anyways.

How would idle time be determined?

dennisguse, 2020-01-06:
The idle time (i.e., the timeout).  could be defined as by the user (creating a settings).
Thissetting could also be used to disable this functionality.

From an implementation perspective the `TrackRecordingService` themself could to take care of this.

NOTE: Pausing might have the disadvantage that the GPS may stay active (I guess; need to be checked). This would drain the battery...

DevRonin, 2020-05-03:
To determine idle time check speed, if below a certain level (ex. 0.5m/s) for 5 sec, then auto pause.

dennisguse, 2020-05-03:
@DevRonin that is actually done to compute moving time.

dennisguse, 2021-04-07:
I guess this is outdated, please re-open if still valid.

closed 2021-04-07 by dennisguse