OpenTracksApp/OpenTracks #118: Additional Metrics

nthh, 2020-01-06:
This fits in with additional support for activity types, but it would be useful to have additional metrics (usually based on the context of the activity)

Examples:
Calories burnt
Skiing - # of jumps, air time, # of runs, slope time, rest time, time on lift

dennisguse, 2020-01-07:
Interesting - if you have any idea how to implement this, it would be probably beneficial.
About time on a lift: By default, the only option would be to pause (before on the lift) and continue.
In a real simple implementation, we could just add a counter via a button (that the user needs to be
press himself); then users can found whatever they want to count.

About calories: this was actually implemented and I removed it.
My pain point is that these estimations are kind of useless as they do not account for anything (basically time, distance, weight) and also do not provide a real benefit (except gamification).
In addition, it requires to store at least the user's weight.
So, I am strongly against this...

DevRonin, 2020-05-11:
Benefit would be .
Benefits to calculate total energy consumption would be for calories counting diets that include physical activity.
Also  would provide insight for activity intensity (energy over time)
Yes, for calculation app need to store weight, age maybe, physical training level.

# of jumps, air time is calculated by garmin devices even for cycling.
for this can't see real benefits :)

dennisguse, 2020-05-11:
@DevRonin I intentionally removed calorie calculation as it is highly useless in terms of precision. And also the calorie intake does not really mean that your body is really processing the calories - depending on the digestion, time of day etc.
From my perspective that is just one of those numbers that doesn't tell you much, but gives you a good feeling.

dennisguse, 2020-05-13:
Don't know what to do....

closed 2020-05-13 by dennisguse