OpenTracksApp/OpenTracks #120: track export creates duplicate files

guzzisti, 2020-01-07:
**Describe the bug**
When doing an export, the app will create duplicate files in the destination dir when there are files from a previous export, for example:
MyTrack.gpx
MyTrack (1).gpx
MyTrack (2).gpx

**To Reproduce**
1. Go to 'Settings'
2. Click on 'export all' then select the desired format
3. Click again on 'export all' and select the same export format as before
4. Files are duplicated

**Technical information**
 - Device: Samsung Galaxy A3
 - OS: Android 8.0 Samsung Stock Rom
 - Version 3.3.1

dennisguse, 2020-01-07:
That is actually a feature - but I am also not in favor of it.
Especially with the enhancement that the folder needs to be selected (due to use of Storage Access Framework); this functionality could be removed.

The responsible code is in `FileUtils`.

guzzisti, 2020-01-07:
Hi, i had a look at this in the `dev` branch and realized that the last time i did an export was when i had version 3.2.11 installed.   
The current version does not use `FileUtils.buildUniqueFileName` at all for the export (its only used in `createTakePictureIntent`) but uses the internal track id as filename instead.

So this issue is obsolete with version 3.3 and above. Sorry for the confusion. :)

closed 2020-01-07 by guzzisti

dennisguse, 2020-01-07:
I was confused myself -  I didn't remember that I removed this "feature" for exports ;)

dennisguse, 2020-01-11:
I just tested the behavior and it was still there (adding a suffix).
`DocumentFile.createFile(displayname)` creates a new file and adds a suffix if the diplayname is already taken.

Fixed in https://github.com/OpenTracksApp/OpenTracks/commit/dd6a4cd153534966ad1340fdb4213341f9923b7f

Is in release v3.3.2