OpenTracksApp/OpenTracks #121: The Android app icon is smallish

licaon-kter, 2020-01-07:
...in all launchers except those based on the Google Pixel one.

Eg. For these https://search.f-droid.org/?q=launcher&page=1&lang=en only Rootless Pixel Launcher and Lawnchair have it full size, all the others (KISS, BlissLauncher, Essential Launcher, Simple App Launcher, etc) show it at 25% or thereabout

[image]

**Technical information**
 - Device: Xiaomi
 - OS:  MIUI11 Android 10
 - Version: 3.3.1 F-Droid

Same as https://github.com/bilde2910/Hauk/issues/52 I guess, maybe the fixes applied there can help.

dennisguse, 2020-01-07:
I cannot reproduce this behavior on a LineageOS 14.1 nor a LineageOS 16.0 with none of the launchers.

I just added the change made by https://github.com/bilde2910/Hauk/issues/52.
Could you test if this fixes the icon size for you?

I only changed the icon that is used for the debug version (it is black then):
https://github.com/OpenTracksApp/OpenTracks/tree/icon

licaon-kter, 2020-01-20:
No joy :(

dennisguse, 2020-01-20:
Then I don't know what to do next.
@licaon-kter if you could try find a solution.
What I don't want to do is adding PNGs - I prefer vector graphics.

dennisguse, 2020-01-22:
I can confirm this issue: tested on another Xioami device using the default launcher.
Could it be that the Android operating system behaves different on Xioami devices encountering vector icons for the launcher?

licaon-kter, 2020-01-22:
I don't know, was that on Android 10 or 9 or?

lennartschoch, 2020-01-22:
Hey there,
I was able to reproduce this behaviour together with @dennisguse on my Xiaomi Mi Mix 3 (MIUI 11.0.4 by xiaomi.eu, based on Android 9). To make sure this issue actually originates from the MIUI launcher, I installed the Pixel launcher and had a look at the launcher icon there:

[image]

Yup, everything's fine there. Seems like the MIUI launcher is indeed doing something weird to the SVG.

dennisguse, 2020-06-05:
@lennartschoch Could you check again?
I might have found the issue: an actual invalid logo size.
Standard defines 108dp instead of 24dp; actually even 48dp up to Android 7.1.

More details here: https://developer.android.com/guide/practices/ui_guidelines/icon_design_adaptive?hl=en

https://github.com/OpenTracksApp/OpenTracks/tree/logoSize%23121

[OpenTracksUpstream-releasePlayStore.zip](https://github.com/OpenTracksApp/OpenTracks/files/4738562/OpenTracksUpstream-releasePlayStore.zip)

lennartschoch, 2020-06-06:
@dennisguse this seems to have done the trick! :)

Before:
[image]

After:
[image]

dennisguse, 2020-06-06:
@lennartschoch Thx!
We will merge the fix ;)

commit 2b8fe4aa (2020-06-07): App icon: be complaint with Android requirements.
Fixes #121.
commit dc7daa6e (2020-06-09): App icon: be complaint with Android requirements. (#253)

Fixes #121.
closed 2020-06-06 by dennisguse