OpenTracksApp/OpenTracks #124: Recording stops when phone goes to sleep

marcocamma, 2020-01-16:
Hi,
thanks for all your work. nice app !
I am running in some issues: with the phone in my pocket the recording stops after a while (few minutes). It might be the Miui11 fault but it does not happen with other apps (like Strava) ... 
I can do some test if you have any suggestion,
thanks

dennisguse, 2020-01-16:
What version of OpenTracks and Android are you using?
There was an issue with Android10 that was fixed. I don't know which version contaons the fix ;)

marcocamma, 2020-01-16:
android 10 QKQ1.19
opentracks 3.3.1

dennisguse, 2020-01-16:
So, I just checked the issue tracker and the previous issue is: #61.
Android 10 requires an additional flag in the manifest allowing a foreground service to record the location.
This was released with v3.2.5.

But it might also be something else.
Recently, I refactored the `TrackRecordingService` and I hope that I did not broke anything.
However, I cannot reproduce the issue :(

@marcocamma Would you be able to investigate this further?
My issue is that I don't have a device available, which makes it almost impossible.

marcocamma, 2020-01-16:
I will try to do some more testing tomorrow or during the weekend
Thanks a lot for looking into this

dennisguse, 2020-01-17:
What you could check:
* just to be sure did you allow the app to record GPS data (in foreground mode)? 
  See: https://developer.android.com/training/location/receive-location-updates
  NOTE: This happens while first starting the app; the app should close if GPS access is denied.
  But does your phone over more options than allow/deny?
* are there some battery saving options?
  (from the OS)
* what happens if you use the "keep screen on feature" (see settings)?
  (you could also try the OS feature for this)
* what happens if you (in addtion) "show on lockscreen feature" (see settings)?
* if you have heart rate monitor: does it work as expected?
  At least it should not reconnect when turning on the screen again...
  NOTE: heart rate data is only recorded if a new GPS point arrives while the UI shows the most recent value.

Additional information:
* just to be sure: do you see a notification while recording (shows the GPS accuracy)?
*

dennisguse, 2020-02-03:
@marcocamma Any more insights?

marcocamma, 2020-02-03:
Ops.... I forgot to report back. Sorry it has been a busy period. It turns
out that MIUI battery saver is the culprit. Changing from 'battery saver
(recommend)' to 'no restrictions' allows data to be recorded. No setting in
opentracks could achieve the same result.
Thanks very much for your work and help
Marco

closed 2020-02-03 by dennisguse