OpenTracksApp/OpenTracks #125: the localisation stops when I take photo

Bebarr, 2020-01-18:
Hi!
I'm with android 8 with the 3.3.1v open tracks version. When I use the app, I disable all stuff that save battery. But when I'm recording a running session and when I take video in the same time, the localisation gone
[image]

I'm sorry if I make mistakes with my English.

dennisguse, 2020-01-18:
Thanks for the report.
Could you provide some more details?
And in the best case a step by step instruction?

Question:
* Did you pause the recording in OpenTracks?
* Did you take a picture or a video or both?
* Did you use another app? Or did you took a picture using OpenTracks (adding a marker)?
* Was OpenTracks in the background?
* Was the track-recording notification shown all the time?

Bebarr, 2020-01-18:
Did you pause the recording in OpenTracks?
I didn't pause the app.

    Did you take a picture or a video or both?
I took pictures and video

    Did you use another app? Or did you took a picture using OpenTracks (adding a marker)?
I took pictures/videos while opentracks runnig but not adding any markers.
When the localisation is on, the App SatSat is automatically activated.
I used also WhatsApp to make and send videos and audio.

    Was OpenTracks in the background?
Yes it was in background all time. And when I put the smartphone in pocket, I lock it.
Also, I enable "show statistics on lock screen"
[image]

    Was the track-recording notification shown all the time?
Yes, and is also shown when the recording is over.

I hope I gave you enough informations :)

Your app is very interesting to me because it use less energy than the osmand~ recording.

dennisguse, 2020-01-18:
Thanks for the information - I will try to reproduce.
At the moment, I am not yet sure what the underlying issue might be.
I will just try to reproduce.

About the energy consumption: good point (never thought about this).
I guess the reason is the following: while the actual GPS recording is similar in OpenTracks and OsmAND~, the latter needs to render a map.

@Bebarr 
Your English is fine. Don't worry ;)
Most of the world population are non-native English speaker (me included).

dennisguse, 2020-01-18:
I can reproduce your issue.
v3.3.2 on a LineageOS 16

1. OpenTracks: start recording
... move
2. open standard camera app and record a video
... move
3. stop video recording
4. OpenTracks: stop recording
5. Export track and open in OsmAnd~

The track should always contain two lines (I ran the same round twice).
However, exactly the part where the camera was recording is not in OpenTracks.
[image]

TODO: Test what happens if a picture is taken (leaving the camera open while moving).

I have absolutely no clue why.

dennisguse, 2020-01-20:
I yesterday also tried this a Pixel 2 (Android 10; default Google version): I could not reproduce this behavior using the default camera app.

And I still have no clue why this would happen.

Bebarr, 2020-01-21:
Ok. I have no idea too. I can't help 😕

dennisguse, 2020-01-21:
@Bebarr I hope somebody else might have a hint...

dennisguse, 2021-04-07:
@Bebarr Please re-open, if this is still an issue for you!

closed 2021-04-07 by dennisguse