OpenTracksApp/OpenTracks #128: Нет подключения к GPS

MrS1R1US, 2020-01-21:
**Describe the bug**
Not connected to GPS from app, Asus ZenFone MaxPro M2, while GPS on the phone is turned on.

**To Reproduce**
1. Go to App
2. Click on GPS
3. Open Status Bar
4. See "GPS launch.." or "GPS connect.."

**Technical information**
 - Device: Asus ZenFone MaxPro M2
 - OS: Android 9
 - Version V3.3.1

dennisguse, 2020-01-21:
Thx. That is more an enhancement as there is no feedback for the user. You need to start a recording before the notifaction is updated...

MrS1R1US, 2020-01-21:
> 
> 
> Thx. That is more an enhancement as there is no feedback for the user. You need to start a recording before the notifaction is updated...

I can also start recording, but GPS still does not work and does not count the distance traveled

dennisguse, 2020-01-21:
Sorry, I misinterpreted your report.
Looks as if the problem is bigger than I expected.
Do you have anymore information?
Is the notification shown while recording GPS?

dennisguse, 2020-01-22:
Another thought: did you change in the preferences the 'GPS accuracy'?
OpenTracks basically ignores all GPS measurements that are beyond this setting.

MrS1R1US, 2020-01-22:
> 
> 
> Another thought: did you change in the preferences the 'GPS accuracy'?
> OpenTracks basically ignores all GPS measurements that are beyond this setting.

history location google - OFF
Send location ELS - ON
GPS accuracy - ON

dennisguse, 2020-01-22:
@MrS1R1US Sorry, I meant the setting within _OpenTracks_.
Did you send change something there (i.e., GPS accuracy)?

MrS1R1US, 2020-01-22:
> @MrS1R1US Sorry, I meant the setting within _OpenTracks_.
> Did you send change something there (i.e., GPS accuracy)?

yes, i try different:
50 m (Recommended) (default)
10 m (Perfect signal GPS)

MrS1R1US, 2020-01-22:
I reset the default settings, nothing helps ..

dennisguse, 2020-01-22:
Could you try to set the  _GPS accuracy_ threshold in the OpenTracks settings to the maximal value (2000m)?

MrS1R1US, 2020-01-26:
I tried, it did not help. Also tried reinstalling the application - unsuccessfully. Kernel 4.4.153-perf+

dennisguse, 2020-01-27:
Then I don't have any further options.
You could try downgrading to an earlier version (F-Droid still has 3.2.11).
One late guess: could it be that your GPS (driver) does not provide a timestamp?
I just remember that I removed a workaround for this in v3.3.X.

MrS1R1US, 2020-01-27:
I tried 2 versions below - it does not work. When you try to add a mark on the map during recording, it says "No GPS signal." At the same time, such applications as Strava, Runtastic and other proprietary ones work, as well as Google Maps, OsmMaps. I asked a friend, on his Xiaomi note 5 - it works. Maybe in the application you can somehow see the logs, why there is no GPS signal?

dennisguse, 2020-01-28:
Do you know how to compile and start the app yourself?
Then you could start it in debug mode and see what happens. The code that you are looking for is in the TrackRecordingService.

MrS1R1US, 2020-01-28:
Unfortunately, I do not have the ability to compile.

dennisguse, 2020-01-29:
Ok, then one last resort: could you try check if other open-source apps (without Google Play services) are working?

https://f-droid.org/en/packages/org.runnerup.free/

https://f-droid.org/en/packages/de.tadris.fitness/

MrS1R1US, 2020-01-30:
amm .... For some reason no fitness app from f-droid works with GPS. But OsmAnd from f-droid works fine with GPS.

MrS1R1US, 2020-01-30:
Do not know what could be the reason? Original firmware from Asus. Lineage OS while is not yet installed because of the guarantee ..

dennisguse, 2020-01-30:
Thanks for testing - that is at least a starting point.
This means that OSMAnd is somehow differently accessing the GPS.

For devs: The interesting part is that all apps in F-Droid cannot use Google Play services and thus also OSMAnd cannot.
Let's check how they are accessing the LocationManager.

MrS1R1US, 2020-01-30:
So what should I do? Forget about OpenTracks?

pstorch, 2020-01-30:
Looks like they do some special things in OsmAnd to make this happen: https://github.com/osmandapp/Osmand/blob/889c15a64a2b1fbc1f10396290106b6d22c279bb/OsmAnd/src/net/osmand/plus/OsmAndLocationProvider.java#L242

dennisguse, 2020-01-31:
If I see it correctly OSMAnd tries to register for GPS __and__ for all other available location providers (network based etc.) that are available.
This indicates that the phone is not providing a location via GPS but via some other source (at least the other apps are not working not implementing this workaround).

@MrS1R1US That means there is nothing we can do about (I am sorry about this).
In OpenTracks, we cannot use the OSMAnd approach as this would on other device results in locations that we should not use as those are not as precise as GPS-based locations.
I guess you would have to report this issue to the device manufacturer and hope for the best.

This will not solve the issue but how does [GPSTest](https://f-droid.org/en/packages/com.android.gpstest.osmdroid/)?
I guess, we should suggests this in the 'help' section for testing of GPS-related issues.

MrS1R1US, 2020-01-31:
I think that after a while I will install Lineage OS and the problem will be solved. But lastly, I would like to suggest that you upgrade OpenTracks, add a pulse record with graphs in final result, the body’s hydration, as well as the height of the ups and downs. You can see visual improvements in Adidas Running (Runtastic), this will completely replace the proprietary application.

MrS1R1US, 2020-01-31:
Look this screenshot: [https://i.imgur.com/h6axHKX.png](url)

dennisguse, 2020-02-03:
@MrS1R1US Can you create separate issues for your ideas?
It is hard to keep track within one ticket.

Elevation gain is covered in #10.
And a graph for heart rate data is already implemented.

closed 2020-02-03 by dennisguse