OpenTracksApp/OpenTracks #129: Unable to uninstall

mmjoshi, 2020-01-21:
I am unable to uninstall this app on Android 10. I am using Nokia 6.1. When I try to uninstall it for the first time, nothing happens. When I uninstall it the second time, get an error,  "Package installer keeps stopping". How do I uninstall?

dennisguse, 2020-01-21:
This Android 10 feature was activated in v3.3.1

mmjoshi, 2020-01-21:
I have v3.3.1. Still unable to uninstall it.

dennisguse, 2020-01-21:
I meant the other way around: your phone does not seem to have this functionality available correctly.
Thus, you cannot uninstall v3.3.1.
What you could do is trying to downgrade to v3.3.0.

Sorry, that this is such a hassle.

mmjoshi, 2020-01-21:
Well, that's not true. I have installed many apps F-Droid along with OpenTracks app which I am able to uninstall. Facing problem only with OpenTracks.
-- 

Sent from my mobile.

mmjoshi, 2020-01-21:
I cleared all the data. Still the same problem. Also unable to downgrade. Get "Failed to install due to unknown error" message. Trying to uninstall because it does not give me the elevation data.  Also, the OsmAnd map is inaccurate.

mmjoshi, 2020-01-21:
Also, am able uninstall other apps from F-Droid. Have problem only with OpenTracks.

dennisguse, 2020-01-22:
@mmjoshi Sorry to hear. Sadly, there is not that much more that I can do - the installer is part of the operating system - what you could try is to capture the logcat output (it might contain some helpful information).

The only option for me would to push a release to F-Droid without `android:hasFragileUserData="true" `.
However, I don't want to disable this feature as nowadays OpenTracks stores all it's data in app-private storage (incl. pictures).
In addition, I cannot build the F-Droid release manually as I cannot sign it (so I cannot craft a special build for you).

So, basically it is a "cannot fix" from my side...

dennisguse, 2020-01-22:
For sake of the issue, you could try to install https://github.com/savoirfairelinux/ring-client-android from F-Droid.
It also has this ` android:hasFragileUserData="true" `.
However, if the issue is due to this, you will have another app that cannot be uninstalled.

And can you disable OpenTracks via system settings?

mmjoshi, 2020-01-22:
Bummer. Does that mean that the problem is with android 10? Hopefully, it gets fixed soon.
-- 

Sent from my mobile.

mmjoshi, 2020-01-22:
Will live with this problem. Won't want to have another app which potentially can not be uninstalled.

dennisguse, 2020-05-20:
@mmjoshi Out of curiosity: could you finally uninstall OpenTracks?

mmjoshi, 2020-05-20:
Yes, I did. I think I had to use three adb command.

It's the new version significantly better than three one I tried? Could give it try once the lockdown is over and I can start running again.

mj

dennisguse, 2020-05-20:
@mmjoshi It definitively is better  - what else can I say :)
Did you have any other issues?

dennisguse, 2020-06-07:
i have the same issue on a Lineage OS 17.1 (GT-I9505).

```
2020-06-07 14:56:31.523 21567-21567/com.android.packageinstaller E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.android.packageinstaller, PID: 21567
    java.lang.RuntimeException: Unable to start activity ComponentInfo{com.android.packageinstaller/com.android.packageinstaller.UninstallerActivity}: java.lang.IllegalArgumentException: Invalid UUID string: 8E17-1CDD
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3270)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3409)
        at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:83)
        at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
        at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2016)
        at android.os.Handler.dispatchMessage(Handler.java:107)
        at android.os.Looper.loop(Looper.java:214)
        at android.app.ActivityThread.main(ActivityThread.java:7356)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:491)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:930)
     Caused by: java.lang.IllegalArgumentException: Invalid UUID string: 8E17-1CDD
        at java.util.UUID.fromString(UUID.java:194)
        at android.os.storage.StorageManager.convert(StorageManager.java:2290)
        at com.android.packageinstaller.handheld.UninstallAlertDialogFragment.getAppDataSizeForUser(UninstallAlertDialogFragment.java:78)
        at com.android.packageinstaller.handheld.UninstallAlertDialogFragment.getAppDataSize(UninstallAlertDialogFragment.java:114)
        at com.android.packageinstaller.handheld.UninstallAlertDialogFragment.onCreateDialog(UninstallAlertDialogFragment.java:179)
        at android.app.DialogFragment.onGetLayoutInflater(DialogFragment.java:417)
        at android.app.Fragment.performGetLayoutInflater(Fragment.java:1351)
        at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1303)
        at android.app.FragmentManagerImpl.addAddedFragments(FragmentManager.java:2431)
        at android.app.FragmentManagerImpl.executeOpsTogether(FragmentManager.java:2210)
        at android.app.FragmentManagerImpl.removeRedundantOperationsAndExecute(FragmentManager.java:2166)
        at android.app.FragmentManagerImpl.execPendingActions(FragmentManager.java:2067)
        at android.app.FragmentManagerImpl.dispatchMoveToState(FragmentManager.java:3057)
        at android.app.FragmentManagerImpl.dispatchActivityCreated(FragmentManager.java:3004)
        at android.app.FragmentController.dispatchActivityCreated(FragmentController.java:184)
        at android.app.Activity.performCreate(Activity.java:7831)
        at android.app.Activity.performCreate(Activity.java:7813)
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1306)
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3245
```

oksjd, 2020-06-21:
Seems to have been fixed on LineageOS 17.1, build 20200616
[image]

dennisguse, 2020-06-21:
@oksjd Thanks for testing!
I also checked and it works now!

PS: Screenshots in English would be great (if it is not too much of a hassle).
We are not all fluent in German and it is just more inclusive I guess.

closed 2020-06-21 by dennisguse

aha999, 2021-04-09:
Lenovo P2, LineageOS 17.1 (Android 10), OpenTracks 3.16.0
OpenTracks can be updated, but not uninstalled, i'm also unable to uninstall [ring-client-android](https://github.com/savoirfairelinux/ring-client-android) that you said we should test for this issue from your [comment](https://github.com/OpenTracksApp/OpenTracks/issues/129#issuecomment-577038965).

dennisguse, 2021-04-09:
@aha999 Can you compile yourself and try if removing `android:hasFragileUserData="true"` from the AndroidManifest helps?
We do not really need this for production release, but I need this for local test builds. So we could also remove it for production.