OpenTracksApp/OpenTracks #130: Uncorrect name when using "Share" button

PhiPhi84120, 2020-01-22:
**Describe the bug**

It's not really a bug, It's a "malfunction". In parameters, I can choose the default name for recorded tracks. I've chosen Date(format ISO 8601). I'm using the "share" option to save the track in the cloud, but the name of the track is a number (a counter) and not the date.

**To Reproduce**
1. Go to Parameters
2. in Recording section, for Default Track name choose Date(ISO 8601)
3. record a track, or select a track recorded
4. Click on 'Share' button
5. Choose a cloud location (in my case it's Nextcloud)
6. The track is correctly saved but it's name is a number and not the date of the track.

**Technical information**
 - Device: ASUS Z00AD
 - OS: Android 6.0.1
 - Version WW_4.21.40.352

dennisguse, 2020-01-22:
Valid point - the export (share is a special case of export) at the moment just uses the (internal) trackId.

For sharing the actual name might be used.
However, than the problem is that the track name needs to sanitized (as it might contain invalid characters).
Code for this is actually (still) in place in `FileUtils`.

contrapunctus-1, 2020-02-03:
Can we please have this for exported tracks too?

dennisguse, 2020-02-03:
@contrapunctus-1 
About exported tracks: the filename consists only of the internal used `trackId` - so if export is triggered twice (tracks were already exported into the same folder) the previously exported files are overwritten (this actually a rather new feature v3.3.2).
We can easily make the name of the track part of the file name - however, if it is changed by the user, some logic is needed to determine the file to overwrite.
A solution might be to use the following format: `trackId_trackName`.
Pull requests are welcome.

dennisguse, 2020-02-14:
Actual simple solution is to change ShareContentProvider to use the following url-path: PREFIX/((trackId))/((track name)).kmz

For exporting: we can also just revert back to the trackname.suffix approach.

PhiPhi84120, 2020-02-14:
Excellent ID !

dennisguse, 2020-03-13:
In the upcoming version sharing files will use the track name.
File name cleanup needs to be done by the app to share.

PhiPhi84120, 2020-03-13:
what a good news! many thanks for your work.

commit ac528f9a (2020-03-16): Use trackname for sharing files.
Fixes #130.
closed 2020-03-16 by dennisguse