OpenTracksApp/OpenTracks #134: Report lap-times when recording and afterwards when viewing workout.

slackline, 2020-01-27:
It would be useful to have lap-times reported both whilst recording and in viewing recorded tracks.

The notification on the defined distance (e.g. 1km) states total time, total distance and pace, but its not clear to me if the pace is for the last 1km or overall and whilst I can do some quick maths to work out my lap time after a few km I tend to slip.  Having the lap-time reported on the distance (i.e. 5m26sec) would be useful to know if pace is being maintained.

Post activity having a breakdown of statistics (time and pace) in a table for each unit distance covered would be informative to, allowing fastest and slowest km (or other user defined distance) to be displayed.

dennisguse, 2020-01-29:
Just to clarify: do you figured what kind of pace (overall vs. last 1km)?
I am not sure what is reported at the moment, I just hope it is the most recent one.
Any insights on this (I don't have the time to check the code :|)?

About the statistics: if don't use [OSMDashboard](https://github.com/OpenTracksApp/OSMDashboard) and OSMAnd~ is installed, you can actually see detailed statistics there.
Actually, https://github.com/OpenTracksApp/OpenTracks/issues/112 is a similar request.

slackline, 2020-01-29:
I have a feeling, but no hard evidence, that currently you get the pace reported for the total distance covered.  This is based on doing a regular run and paying attention to the times on the 1km laps because my run starts with a short uphill then relatively flat for 5km, after that its gradual uphill and then it steepens and I'm sure I slow down on those sections but the pace reported only drops gradually (if it were reported as the pace per 1km lap then 6km and 7km would be really low).  At the end of the run its spot on for the total distance (~11km in 1hr2mins).

Hadn't come across OSMDashboard before but will check it out, thank you.

#112 is interesting too (I had actually read it before submitting this issue), but its really the reporting of the pace whilst recording out on a run/cycle that I find appealing.  I keep note mentally of the last total distance thats reported (i.e. "you have done 5km in 26m and 37sec at a pace of 10.9km/h") and when I hit the 6km distance work out how long its taken me to do the 6th km, but I would find it more useful having "your lap time is 5m 14sec" reported (for every 1km or other unit as specified) rather than the overall pace.

dennisguse, 2020-01-29:
Understood :)

Just one tiny but: if you time to implement this please do so (I won't be able to invest much time in OpenTracks until mid-March).
I guess it is not too difficult.
@pstorch would be able accept pull requests (I hope).

slackline, 2020-01-29:
I'll have a look, but know very, very little Java (only started learning Python a year and a half ago due to job change, previously knew basic R as I'm a statistician by trade).

Chances are by the time I can understand how it works it will be May!  But I'll have a look and see if I can work it out.

dennisguse, 2020-01-29:
@slackline don't worry.
The place you are looking for is here: https://github.com/OpenTracksApp/OpenTracks/blob/df502f835c585433bea6735e21bfa5c8b61a8be0/src/main/java/de/dennisguse/opentracks/services/tasks/AnnouncementPeriodicTask.java#L200

slackline, 2020-01-29:
Thanks for the pointer, just cloned your repo and will make a branch and see if I can work it out.

dennisguse, 2020-03-14:
I just checked the code: currently the "total" pace is reported.

dennisguse, 2020-03-14:
Actually, reported is at the moment the following: total distance, total time, and total speed/total pace.
With regard to current pace/speed: this is not directly possible as this data is not part of  `TripStatistics` and thus not passed to `AnnouncementUtils`.

With regard to lap time: similar to current pace/speed, but would in addition require to compute split intervals (incl. configuring split intervals).

slackline, 2020-03-15:
Ok, thanks for taking a look into this and reporting back.

I get a rough idea of lap times from the total pace and can live with that.

commit a6d55664 (2020-08-28): Announcement: while recording report current interval pace or speed.
Fixes #134.
commit 5be4c107 (2020-09-02): Announcement: while recording report current interval pace or speed.
Fixes #134.
closed 2020-09-02 by rgmf