OpenTracksApp/OpenTracks #135: Pause a record long time stop recording

pieplu, 2020-01-30:
**Describe the bug**
When record is paused a long time, it automaticly stop, and we are not able to continue recording on the same track.

**To Reproduce**
1. Start a record
2. Click on pause
3. Wait... maybe one hour
4.Come back en the app, and try to continue recording

**Technical information**
 - Device: Xiaomi Mi A2 lite
 - OS: Android
 - Version 9

dennisguse, 2020-01-30:
Thanks for the report - I did not yet try to reproduce it (I did not want to wait an hour), bu there is a way to reproduce this:
1. Start a record
2. Click on pause
3. Open Android task switcher and swipe activity up
4. Open OpenTracks again
-> Recording is finished (and cannot be continued due to #11)

The explanation is rather simple: OpenTrack got terminated (really closed) by the operating system.
This also includes the `TrackRecordingService` (notification is removed).
And there is no continue feature.

The question is: why does this happening as doing the same (except pausing) does close the `TrackDetailActivity`, but keeps the `TrackRecordingService` alive (notification still there).

A fix might be possible here: https://github.com/OpenTracksApp/OpenTracks/blob/df502f835c585433bea6735e21bfa5c8b61a8be0/src/main/java/de/dennisguse/opentracks/services/TrackRecordingService.java#L717

In case of an ongoing recording (and then pausing), the `TrackRecordingService` leaves the foreground mode and thus can be killed by the android operating system (e.g., due to low memory).

closed 2020-02-03 by dennisguse

pieplu, 2020-02-03:
Thanks @dennisguse :)

PR #142 (2020-02-03): Fixes
Fixes #135 and #109.