OpenTracksApp/OpenTracks #143: Executing tests deletes all data

dennisguse, 2020-02-03:
As observed in https://github.com/OpenTracksApp/OpenTracks/pull/133#issuecomment-581637594 it is a problem that the tests delete all data (when executed on a real device).

This happens in: 
* `TrackRecordingServiceTest`
* `TrackRecordingServiceTestLooper`

pstorch, 2020-02-03:
Is it possible to use a different database name for unit tests?

pstorch, 2020-02-03:
I executed `ContentProviderUtilsTest`

dennisguse, 2020-02-03:
Something like this might work: https://stackoverflow.com/questions/9804917/android-testing-start-with-clean-database-for-every-test/38247496

But I am out for today :)

ristello, 2021-10-04:
Hi,
is this issue still relevant?
I executed the instrumented tests on my phone. The tracks in the DB of the app had not been changed.
It seems that the instrumented tests only change the DB of the debug-app.

dennisguse, 2021-10-04:
@ristello exactly :)
I usually use the debug build and then it is quite annoying that the data is gone.
But it is not the most important thing todo.

ristello, 2021-10-12:
@dennisguse I have filed a PR which would solve that issue. The approach is described in the PR.

dennisguse, 2021-10-16:
This is fixed - thanks to @ristello!

commit 3327599d (2020-07-27): SearchEngineTest: do not use production database.
Partly fixes #143.
commit b676fd66 (2020-08-10): SearchEngineTest: do not use production database.
Partly fixes #143.