OpenTracksApp/OpenTracks #148: Remove "elevation min/max" completely

dennisguse, 2020-02-14:
At least while recording, the elevation min/max is not really helpful.
Therefore, I would like to remove it.
In addition, it is highly unprecise as the GPS is used.
Often elevation min is recorded directly after starting a track as the GPS does not yet have a good fix.

dennisguse, 2020-02-14:
Also change the label in the settings "elevation/grade" to elevation.

rgmf, 2020-02-21:
GPS elevation is inaccurate but for me it is useful. For example, when I run through the mountain it helps me to know where the track is where I have run in a higher situation. I know it's inaccurate, but it helps me get some statistics. The application could inform the user about the inaccurate elevation. what do you think about it?

dennisguse, 2020-02-21:
@rgmf This is possible from Android 8 onwards (if I remember correctly): see #78.

What this issue is about is to remove the min and max elevation as it is (in my opinion) not really useful (and can also be derived from the charts).
But I will keep the absolute elevation (derived from GPS in place).

rgmf, 2020-02-21:
Ok, I understand. It's fine ;)

commit 92dfadfd (2020-03-16): Remove elevation min/max from UI as GPS-derived data often lead to invalid data (especially initial fix phase).
Fixes #148.
closed 2020-03-16 by dennisguse