OpenTracksApp/OpenTracks #149: skateboarding/longboarding activity type

sporksmith, 2020-02-14:
**Is your feature request related to a problem? Please describe.**

I use this for longboarding a lot. None of the built in activity types fit.

**Describe the solution you'd like**

Add a "skateboarding" activity type, with an icon.

**Describe alternatives you've considered**

Being able to add custom activity types to the list in general would be nice. I can already type "longboarding" for each individual activity but usually don't bother.

dennisguse, 2020-02-14:
I agree. As a workaround, you can set the default activity to longboarding (but there is no icon).

dennisguse, 2020-02-14:
And #14 is related.

sporksmith, 2020-02-14:
Ah whoops, looks like #14 covers it. Thanks!

closed 2020-02-14 by sporksmith