OpenTracksApp/OpenTracks #151: ability to set preferred rate by activity type

sporksmith, 2020-02-14:
It's currently a global setting but e.g. I prefer cycling in miles/hr but running in minutes/mile.

dennisguse, 2020-02-14:
Is it about the UI or the voice announcements?
#87 is related.

sporksmith, 2020-02-14:
I'm referring to the UI.

I haven't used the voice announcements, but I imagine it'd be applicable there too.

rgmf, 2020-02-21:
What do you think something like that?

In **Settings** you have **Preferred rate** with two options: **Speed or Pace**. Well, you can put a third option: **Automatic** (or something like that). If the:

- User choose automatic the app will use pace for RUN and WALK and speed for the others
- User choose speed, the app will use speed for all tracks.
- User choose pace, the app will use pace for all tracks.

Something like this:
```
diff --git a/src/main/java/de/dennisguse/opentracks/fragments/ChartFragment.java b/src/main/java/de/dennisguse/opentracks/fragments/ChartFragment.java
index bed4b8d8..4035df29 100644
--- a/src/main/java/de/dennisguse/opentracks/fragments/ChartFragment.java
+++ b/src/main/java/de/dennisguse/opentracks/fragments/ChartFragment.java
@@ -79,6 +79,7 @@ public class ChartFragment extends Fragment implements TrackDataListener {
     private boolean metricUnits = true;
     private boolean reportSpeed = true;
     private int recordingDistanceInterval;
+    private String category = "";
 
     // Modes of operation
     private boolean chartByDistance;
@@ -109,7 +110,7 @@ public class ChartFragment extends Fragment implements TrackDataListener {
                 }
             }
             if (PreferencesUtils.isKey(getContext(), R.string.stats_rate_key, key)) {
-                boolean speed = PreferencesUtils.isReportSpeed(getContext());
+                boolean speed = PreferencesUtils.isReportSpeed(getContext(), category);
                 if (reportSpeed != speed) {
                     reportSpeed = speed;
                     chartView.setReportSpeed(reportSpeed);
@@ -202,8 +203,10 @@ public class ChartFragment extends Fragment implements TrackDataListener {
         if (isResumed()) {
             if (track == null || track.getTripStatistics() == null) {
                 startTime = -1L;
+                category = "";
                 return;
             }
+            category = track.getCategory();
             startTime = track.getTripStatistics().getStartTime();
         }
     }
diff --git a/src/main/java/de/dennisguse/opentracks/fragments/StatsFragment.java b/src/main/java/de/dennisguse/opentracks/fragments/StatsFragment.java
index 427ab630..b01acf37 100644
--- a/src/main/java/de/dennisguse/opentracks/fragments/StatsFragment.java
+++ b/src/main/java/de/dennisguse/opentracks/fragments/StatsFragment.java
@@ -481,7 +481,7 @@ public class StatsFragment extends Fragment implements TrackDataListener {
         String trackIconValue = TrackIconUtils.getIconValue(getContext(), category);
 
         boolean metricUnits = PreferencesUtils.isMetricUnits(getContext());
-        boolean reportSpeed = PreferencesUtils.isReportSpeed(getContext());
+        boolean reportSpeed = PreferencesUtils.isReportSpeed(getContext(), category);
         boolean isRecording = isSelectedTrackRecording();
```

dennisguse, 2020-02-22:
@rgmf Looks like a good approach

dennisguse, 2020-02-22:
@sporksmith for the time being you could install the Fdroid and the PlayStore version in parallel and use one for cycling and one for running.

sporksmith, 2020-02-24:
> In Settings you have Preferred rate with two options: Speed or Pace. Well, you can put a third option: Auto (or something like that)

@rgmf That'd probably be good enough for my purposes. Ideally I'd take it even further and make 'auto' the default, and make the overrides be per-activity-type instead of a global setting. (Though I guess there needs to be some global default for new/custom activity types)

> for the time being you could install the Fdroid and the PlayStore version in parallel and use one for cycling and one for running.

@dennisguse interesting idea, though I think I prefer my current workaround of just flipping the global speed/pace setting back and forth :)

dennisguse, 2020-03-13:
@rgmf Could you make a PR for your proposed change?

rgmf, 2020-03-13:
@dennisguse, let me do what @sparksmith suggested in his comment about making the overrides be per-activity-type instead of a global setting so it be ready for new/custom activity types. Next week I have free time to work on it.

dennisguse, 2020-03-14:
@rgmf I like the idea proposed by @sporksmith.
But at the moment, I would not like to introduce such a feature as it requires a lot of work with limited benefits.
I like the idea about this "auto"-feature as this can be implemented in a simple way and solves the immediate problem.

commit 83b5ebb0 (2020-04-27): Added new option in Preferred rate setting: By Default.
With By Default option it will use pace or speed depending on activity type.
Fixes #151.
closed 2020-04-27 by rgmf