OpenTracksApp/OpenTracks #152: share location by distance, time or inactivity

nutpantz, 2020-02-15:
On a recent hike, i got worried that i might be in trouble. I started sending location texts every 30 minutes just in case i had serious issues. And i always had to exit open tracks to do so.
It would be nice to have an automatic feature that could send your location repeatedly on a schedule or if you stopped moving (not if you lost gps signal) for a set amount of time. (Email or sms message) And am alarm to let you know the app plans to send your location before hand.
A repeating 3 noise emergency  alarm when it thinks your in trouble every 15 minutes might be helpful also.

dennisguse, 2020-02-16:
That is a very specific feature request. I guess it is out of scope for OpenTracks. You can check Seguime: https://f-droid.org/app/pc.javier.seguime

Using both apps in parallel should not add to much battery drain as the GPS is shared by both.

closed 2020-02-16 by dennisguse

nutpantz, 2020-02-16:
Sadly that seems to require internet which is not always available or cheap compared to sms world wide

dennisguse, 2020-02-17:
Seguime can also send SMS from the phone directly although the feature is quite hidden in the settings.