OpenTracksApp/OpenTracks #153: Add export to relive (gpx or tcx)

Flavien06, 2020-02-16:
Hi, actually export is to kmz.
it is possible to change default format to gpx or tcx.
or option to convert and share in gpx for send to relive app?
thank

dennisguse, 2020-02-16:
What kind of export are you referring to?
I guess the share or the show on map functionality? 

Tcx is not implemented.
I have the feeling it was some time ago and I removed it....

Flavien06, 2020-02-16:
I need an option to export the course in gpx and share it immediately with other application

dennisguse, 2020-02-17:
I see. I put it on the roadmap, but with low priority. Pull Request welcome.

pirovc, 2020-04-06:
If you open the track with OsmAnd you can export the gpx directly

dennisguse, 2020-04-06:
@Flavien06 Could you describe your desired solution?

dennisguse, 2020-05-11:
I guess, #225 is a duplicate but provides a nicer description.

closed 2020-05-11 by dennisguse

Flavien06, 2020-05-11:
> 
> 
> @Flavien06 Could you describe your desired solution?

Actualy the "share" button share .kmz file.
I want this button "share" (export) in gpx.

A general option for chose the output file.
Or a second button to convert .kmz to .gpx and share it.