OpenTracksApp/OpenTracks #155: "By Time" graph shows no data

eliotb, 2020-02-25:
**Describe the bug**
The By Time graph appears to have no data.  The grid and y axis labels seem OK
Possibly there are a bunch of overlapping numbers in the bottom right corner.
Possibly there is a single vertical line on the RHS of the graph. (Hard to see)
The By-Distance graph seems fine.

**To Reproduce**
View By Time tab for any track.

If applicable:
* add screenshots - will try to get
* add logcat output - if requested?

**Technical information**
 - Device: Moto G
 - OS: /e/ based on lineage_falcon 7.1.2, v 0.7-20191119
 - Version 3.4.0 code 3287

eliotb, 2020-02-25:
[image]

dennisguse, 2020-02-26:
I can reproduce this.
I will put it in the roadmap, but I don't have time before mid-March.
Thanks for the report.

dennisguse, 2020-03-13:
I cannot reliablely produce this .
On one device it occurs while on another it doesn't.
Any suggestions?

eliotb, 2020-03-14:
? screen size. Mine is  720x1280?

Comparing your latest [screenshot](https://github.com/OpenTracksApp/OpenTracks/blob/master/fastlane/metadata/android/en-US/phoneScreenshots/screenshot_5.png),
and looking closely at the numbers piled up in the corner of mine:
(Total time 1:52:14, moving time 1:12:29)
Mine seems to start with 00:-60  and end with something...8977  - wierd numbers for times.
I'm in timezone UTC+13 at the moment.

I'm happy to share the kmz file with you privately.

dennisguse, 2020-04-10:
@eliotb v3.6 was just released (F-Droid will take 2-3 days).
There was some refactoring of the charts and I might have accidentally fixed this issue.
I will close the issue now, but if the issue is still valid in v3.6 then please reopen it.

eliotb, 2020-04-20:
Sadly the problem persists in the latest release. AFAICS I cannot reopen this issue myself.

dennisguse, 2020-04-20:
@eliotb Can you send me the track (KML and GPX) to dennis.guse@alumni.tu-berlin.de?
And does this affect all tracks on your device? (Can you try to record a new one?)

eliotb, 2020-04-21:
Done.  Yes it affects all tracks, old and new.  Also affects the graph during recording.

dennisguse, 2020-04-21:
@eliotb Thanks (I got the file).
Just one more question: which version of OpenTracks are running (Settings -> About)?

PS: I will probably check this on the upcoming weekend.

eliotb, 2020-04-21:
On 21/04/20 5:30 pm, Dennis Guse wrote:
> @eliotb <https://github.com/eliotb> Thanks (I got the file).
> Just one more question: which version of OpenTracks are running
> (Settings -> About)?

Before recording to that track I updated to  v3.6.2 code 3335

> 
> PS: I will probably check this on the upcoming weekend.

If you give me a hint about where to look for the graphing code, I will
see if I can spot anything... (I'm no java/android programmer though!)

dennisguse, 2020-04-21:
That's the issue: I am not entirely what caused this... (I don't recall any related changes before v3.6.0).
Either it is the data preprocessing for (actually unlikely as it works on other devices with a higher resolution) or somehow the screen resolution breaks something (Moto G is rather small).
Or the record is broken (that's something I can check).
I saw the same behavior on a Moto G 4G, but cannot reproduce it now.

Anyhow, it should be either in:

https://github.com/OpenTracksApp/OpenTracks/blob/fbc902e83042dd9239d79ebbf9858e48a22f1788/src/main/java/de/dennisguse/opentracks/fragments/ChartFragment.java

https://github.com/OpenTracksApp/OpenTracks/blob/35dec1634cd3815f6d264bf6dee60c10eee3b22a/src/main/java/de/dennisguse/opentracks/chart/ChartView.java

https://github.com/OpenTracksApp/OpenTracks/blob/fbc902e83042dd9239d79ebbf9858e48a22f1788/src/main/java/de/dennisguse/opentracks/chart/ChartValueSeries.java

eliotb, 2020-04-21:
On 21/04/20 5:45 pm, Dennis Guse wrote:
> That's the issue: I am not entirely what caused this... (I don't recall
> any related changes before v3.6.0).
> Either it is the data preprocessing for (actually unlikely as it works
> on other devices with a higher resolution) 

I installed OpenTracks on another phone (Huawei Y5), transferred a track
kmz across - the graph is fine there.
Interestingly the stats are *different* on the 2 devices... Transferred
track distance is bigger, moving time bigger, total time smaller!?

I do understand that this is probably low priority for you, and it is
really a minor irritation for me, so if you want to push it to the back
of the queue I won't complain.

> or somehow the screen
> resolution breaks something (Moto G is rather small).
> Or the record is broken (that's something I can check).

AFAICS the data is OK.  If I send the track to map (OsmAnd), it can
display the data as Altitude+Speed vs (Time or Distance or Time of day).
 All these appear correct.

> I saw the same behavior on a Moto G 4G, but cannot reproduce it now.
>
> Anyhow, it should be either in: [...]

-- 
Eliot

dennisguse, 2020-04-22:
Thanks.
@eliotb Can you try to import the track on the Moto G and check if the issue still occurs?
I also opened the KMZ and the timestamps were somehow weird (year 2000, but the track time was 2020).

About the stats: I created #192 to keep track of this.

eliotb, 2020-04-22:
Well hmmmmmmm. So the imported tracks show the By Time graph correctly on the Moto G, but as you say, the date is in 2000.   E.g. 2020-04-01 became 2000-09-05

dennisguse, 2020-04-22:
Can you try to record a new track and check the times in the exported KML?

dennisguse, 2020-04-22:
The time for TrackPoints is actually taken directly from the GPS and there was rollover of the time counter some month ago (happens every 20 years?). This would explain the old timestamps, which might lead to invalid diagrams.

dennisguse, 2020-04-23:
I did the math: and it looks like GPS time rollover (happening every 1024 weeks) [1].
This leads to the situation that the start time of the track (via `System.currentTimeMillis()`) is far newer than the timestamps reported by your GPS.
From your track:
* Track start date: 2020-04-21
* GPS date: 2000-09-05
Difference is 7168 days (i.e., 1024 weeks).

However, the Chart assumes that the track start time corresponds to the GPS time.
Which in this case doesn't...

So, we have a hardware fault and an invalid assumption :)

[1] https://en.wikipedia.org/wiki/GPS_Week_Number_Rollover

dennisguse, 2020-04-23:
> Well hmmmmmmm. So the imported tracks show the By Time graph correctly on the Moto G, but as you say, the date is in 2000. E.g. 2020-04-01 became 2000-09-05

That is actually expected as the startTime is extracted on import from the first GPS point while the original startTime (from the track) is discarded. See #192.

@eliotb 
I implemented a fix for the GPS week rollover: https://github.com/OpenTracksApp/OpenTracks/tree/gpsTimeRollover
Could you test it on your device?

eliotb, 2020-04-23:
Happy to.  Do you have an apk handy? ;)  
If I have to build it myself, I may be away for a while...

eliotb, 2020-04-27:
Built and tested - Success!

closed 2020-04-27 by dennisguse

dennisguse, 2020-04-27:
@eliotb Cool!
The fix will be in the upcoming release (happening this week).

The tracks that you already recorded would need to be fixed manually:
1. export the tracks (KML if they contain no pictures otherwise KMZ) 
2. open the KML file (for KMZ: it is a zip containing a doc.kml)
3. for all gps data: replace the date with the date from the header
4. import into OpenTracks
5. delete old (broken) tracks

eliotb, 2020-04-28:
Thanks Dennis.  Basically I already did that by accident trying to export and import one file, instead doing them all ;) I look forward to the new release.
Also handy to know that I can compile and install an android app myself.  (Not much skill required, gradle made it easy...)

dennisguse, 2020-04-29:
@eliotb Just be aware that the timstamps for those tracks will something in the year 2000.
If that does not bother you: all fine.

PS: This was a really fun bug report :)

commit e6517942 (2020-04-27): Workaround for GPS week rollover (happening every 19.7 years).
Fixes #155.

https://en.wikipedia.org/wiki/GPS_Week_Number_Rollover
commit b7b7bdf1 (2020-04-27): Workaround for GPS week rollover (happening every 19.7 years).
Fixes #155.

https://en.wikipedia.org/wiki/GPS_Week_Number_Rollover