OpenTracksApp/OpenTracks #159: Use track name when exporting

dennisguse, 2020-03-13:
Follow up to #130.

rgmf, 2020-04-16:
@dennisguse, Why do you think the name of exported files are important? I mean, users only use import/export for take their activities from one place to another or for backup purpose, aren't they?

And now is easy to handle non-duplication tracks when exporting.

dennisguse, 2020-04-17:
It was mentioned here https://github.com/OpenTracksApp/OpenTracks/issues/130#issuecomment-581579996

@contrapunctus-1 Can you elaborate on your requirement that the "export all" function should also use the track name instead of the trackId?

contrapunctus-1, 2020-04-17:
@dennisguse it makes it easier when opening them in a file manager or other applications - you have a better idea of which file is what, without needing to open them.

rgmf, 2020-04-18:
@contrapunctus-1 @dennisguse what do you think about using a name like this?
`<trackId>-<trackName>.<extension>`

For example, a track:
- which id is: 321
- which name is: trail running at Pyrenees

It'll have this name:
`321-trail running at Pyrinees.kmz`

It's easy change the exporting code to not duplicate when exporting looking for the id and the name is descriptive.

contrapunctus-1, 2020-04-18:
I'd suggest including the starting datetime in there (in the ISO 8601 format; OsmAnd does it too).

rgmf, 2020-04-30:
@contrapunctus-1 using a ISO 8601 prefix when exporting if there is a file with the same prefix would be overwritten, isn't it? Because if two track names coincide in ISO 8601 data time are the same track, aren't they?

dennisguse, 2021-04-01:
@rgmf This is fixed, right?

rgmf, 2021-04-01:
@dennisguse Yes, it is.

commit a00b8244 (2020-05-02): Use track start time (ISO 8601) and track name when exporting. Fixes #159.
closed 2021-04-01 by dennisguse