OpenTracksApp/OpenTracks #166: Importing a track creates file:// for attached pictures

dennisguse, 2020-03-16:
**Describe the bug**
Importing a track with images, creates file:/// urls instead as content:/// as photoUrl.

**To Reproduce**
1. Create a track with one marker and photo
2. Export track
3. Import exported track
4. A file:// is created for the photoUrl

Was discovered as a side effect of #162.

Does not affect anything in the system.

dennisguse, 2020-03-16:
I guess the issue occurs here: https://github.com/OpenTracksApp/OpenTracks/blob/df502f835c585433bea6735e21bfa5c8b61a8be0/src/main/java/de/dennisguse/opentracks/io/file/importer/KmzTrackImporter.java#L155

dennisguse, 2020-03-16:
Got it: https://github.com/OpenTracksApp/OpenTracks/blob/bbb6a7c22cf2ebbabdbbd3c38e4e9ff565e4dd68/src/main/java/de/dennisguse/opentracks/io/file/importer/AbstractFileTrackImporter.java#L382

And I found some weird behavior: we actually export the complete content-url.
This needs to be changed....

rgmf, 2020-04-05:
Why is `getPhotoUrl` needed in `AbstractFileTracker`?

   ```java
 protected String getPhotoUrl(String fileName) {
        if (importTrackId == -1L) {
            return null;
        }
        File dir = FileUtils.getPhotoDir(context, importTrackId);
        File file = new File(dir, fileName);
        return Uri.fromFile(file).toString();
    }
```

The only place where `getPhotoUrl` method is used is in `KmlFileTrackImporter` before adding the waypoint to get the photo url. But the photo url already is in `photoUrl` member (like content:///).

dennisguse, 2020-04-05:
Valid point - it is probably something that I overlooked while migrating to Storage Access Framework.
And the weird thing is that the photoUrl within in the KMZ is actually not valid or even useful...
I guess the basic issue is that, we have (very) limited test coverage for all this code, but it was significantly modified.

rgmf, 2020-04-05:
Aaahhh ok, I understand.

> And the weird thing is that the photoUrl within in the KMZ is actually not valid or even useful

Yes, photoUrl is not valid because in kml file the photoUrl has the ID of the track in the source app (exporter) but it won't be the same in the destination (importer). For example, in this kml snippet:

```kml
<PhotoOverlay>
	<name><![CDATA[Marker 0]]></name>
	<description><![CDATA[]]></description>
	<Camera>
               <longitude>-0.805818</longitude><latitude>38.158598</latitude>
               <altitude>20</altitude>
               <heading>-99.95959</heading>
               <tilt>90</tilt>
        </Camera>
	<TimeStamp><when>2020-04-05T10:39:51Z</when></TimeStamp>
	<styleUrl>#waypoint</styleUrl>
	<Icon>
               <href>content://de.dennisguse.opentracks.debug.fileprovider/pictures/Pictures/185/Apr 5, 2020 12_39_49 PM.jpeg</href>
        </Icon>
	...
</PhotoOverlay>
```

Like you can see content:// has the id of this track in the importer in the url (**185**):

_content://de.dennisguse.opentracks.debug.fileprovider/pictures/Pictures/**185**/Apr 5, 2020 12_39_49 PM.jpeg_

If you don't mind I'm going to try to fix it.

dennisguse, 2020-04-05:
@rgmf If you have time - it is a rather ugly bug.

PR #178 (2020-04-10): Importing creates content:// for attached pictures and exporting with…
…out points bug fixed. Fixes #166 and #176.

**Describe the pull request**
"Import all" option now creates content:// for attached pictures instead of file://

Also, "Export all" option doesn't results in a null exception when a file without points is exported.

**Link to the the issue**
#166 
#176 

**License agreement**
By opening this pull request, you are providing your contribution under the _Apache License 2.0_ (see [LICENSE.md](LICENSE.md)).
commit 1b69f297 (2020-04-10): Importing creates content:// for attached pictures. Also, exporting without points bug fixed and images href contains relative link. Fixes #166 and #176.
closed 2020-04-10 by dennisguse