OpenTracksApp/OpenTracks #176: Export all with tracks without points results in a null exception

rgmf, 2020-04-06:
**Describe the bug**

When you use the "Export all" option in Settings the app cracks and exits if there are tracks without points.

This is because in `KmlTrackWriter` `writeBeginTrack` can be called with a null in the `startTrackPoint` argument. Later, in this method `startTrackPoint` is used without care if it's null or not.

Seem easy to fix but like here is writting out a kml file it's needed testing it deeply and with care so the output file fulfil kml standard. The key question here is: Can exists a <Placemark> without <Point>? If so, this bug is easy to fix.

**To Reproduce**
1.- Record a track with GPS disabled. Only total time is saved.
2.- Stop and save the track.
3.- Go to Settins -> Export all.

Was discovered in #166 because I needed tracks to import so I exported all my tracks.

dennisguse, 2020-04-06:
Nice that you found this one.

rgmf, 2020-04-09:
Fix in https://github.com/OpenTracksApp/OpenTracks/pull/178

closed 2020-04-09 by rgmf