OpenTracksApp/OpenTracks #177: App does not track location when device is locked on Android 10

lennartschoch, 2020-04-06:
When app is being run on a device using Android 10, it is unable to obtain the users location whenever at least one of the following two conditions is not given:

- The app is running in the foreground
- The device is unlocked

When I checked the apps permission, I noticed how there were only two possible values for location access:

`Allow only while using this app` and `Deny`.

After I did some investigation, I figured out that there is a [new permission required](https://developer.android.com/training/location/request-updates#request-background-location) from Android 10 on when obtaining a users location in the background. After I added the following line to `AndroidManifest.xml`:

`<uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />`

the default setting when granting location access was still `Allow only while using this app` (for some reason even after reinstalling the app), but I could change this setting to `Allow all the time` manually now. Now, the app was successfully able to obtain a users location when the phone was locked, but the app was still in the foreground. What still is missing is the ability to keep tracking when the app is running in background mode on Android 10.

I will conduct some further investigation and try to find the root cause of the problem.

**Technical information**
 - Device: Xiaomi Mi Mix 3
 - OS: MIUI by xiaomi.eu 11.0.2
 - Version: v3.5.0

dennisguse, 2020-04-06:
IMHO, OpenTracks should not require `ACCESS_BACKGROUND_LOCATION`.
For the recording of location data, a _foreground service_ is started and it is flagged as:
https://github.com/OpenTracksApp/OpenTracks/blob/fbc902e83042dd9239d79ebbf9858e48a22f1788/src/main/AndroidManifest.xml#L125
Then this service is considered a "continued user interaction" [1] and should be able to collect locations while running.
This also requires it to show a notification.

@lennartschoch Do you have a notification while tracking?
And is there some battery optimization enabled?

@lennartschoch As a workaround, you can use the keep screen on while recording option in combination with the show on lock screen option.

[1] https://developer.android.com/training/location/request-updates#continue-user-initiated-action

lennartschoch, 2020-04-06:
That makes sense.

I do have a suspicion about the battery optimizer:
There is a super hidden setting in MIUI under `Settings > Battery & performance >  [Settings icon] > App battery saver > Sleeping apps > OpenTracks` with the following default setting: 
`Identify apps and activities to keep important processes going. Battery saver will treat important apps selectively`.
I changed this setting to `Battery saver doesn't restrict app's activity` and walked for a few hundred meters with the app in the background.

Seems like everything runs smoothly now🎉

I will do the same on a longer route tomorrow and let you know whether that's really the solution.

lennartschoch, 2020-04-07:
Can confirm that disabling the battery saver for OpenTracks solves the issue on MIUI.
Will close the issue now.

closed 2020-04-07 by lennartschoch

dennisguse, 2020-04-07:
@lennartschoch Thanks for testing.
And it was a pretty good bug report!