OpenTracksApp/OpenTracks #180: Improve import/export all tracks

rgmf, 2020-04-09:
### Is your feature request related to a problem? Please describe.
When user imports kmz files with photos there is a potencial problem: if there are waypoints photos then the URL can be invalid and the code doesn't check anything about this now.

This is likelihood because is an external file (it's outside OpenTracks) so we can't do any assert.

For example, you could have this kmz file:
```
9.kmz
  |__ images
        |__ beach.jpeg
  |__ doc.kml
```
And in the `doc.kml` the URL to the image inside images could be wrong, for example:

```kml
<href>images/mountain.jpeg</href>
```

If you import a file like this, the importer doesn't complain but we'll have a broken link and the image won't be show.

### Describe the solution you'd like
**In the importer (critical):**
It's critical because we are importing an external file. Importer should check that the href in the kml file points to a real image inside kmz file structure.

If importer run into with a broken link then it should go on without image.

**In the exporter**
After https://github.com/OpenTracksApp/OpenTracks/pull/178 the exporter creates an href with a relative path to the images inside kmz file.

In the exporter we should be careful as we never now what is actually in the database (and what some previous OpenTracks version actually did).

Now every waypoint in the database with a photoUrl value results in a `<PhotoOverlay>` in the kml file. But it's necessary to check if this photoUrl point to a valid image inside OpenTracks external storage before creates a `<PhotoOverlay>`.

In the database you can find file:// and content:// because https://github.com/OpenTracksApp/OpenTracks/issues/166 issue.

**Attribute photoUrl**
It uses an attribute called `photoUrl` throughout the importer / exporter which can be misleading because it is used to save the `waypoint.photoUrl` and to save the path relative to the photo in the kmz file.

**Tests**
~~Maybe~~ we should add more test.

### Additional context
- See https://github.com/OpenTracksApp/OpenTracks/pull/178 discussion.
- There is another change that affected photoUrl (use of a FileProvider to get images from the camera): 828a19a

dennisguse, 2020-04-09:
@rgmf I just wanted to add a commit to this PR, but I don't have access rights....
I pushed it to `origin` into branch: `rgmf-issue#166`.
It just checks the photoUrl while importing.

dennisguse, 2020-04-09:
One question is actually, if we should consider the KMZ internal file name at all.
We could also generate a one within OpenTracks.

rgmf, 2020-04-09:
> One question is actually, if we should consider the KMZ internal file name at all.
> We could also generate a one within OpenTracks.

We need to consider KMZ internal files. The current code get the name (NOT the complete path but only the name) of the file from kml and then expect an image with the same name inside images folder.

For example, here we have three images related with three waypoints:

```
9.kmz
  |__ images
        |__ beach.jpeg
        |__ river.jpeg
        |__ mountain.jpeg
  |__ doc.kml
```

How do you know which image corresponds to a waypoint? The importer know it through the value of hrefs.

dennisguse, 2020-04-09:
That's true, but I meant the filename as which we store the image within OpenTracks. We don't need to use the one extracted from href.
We could for example use the waypointId.

rgmf, 2020-04-09:
Aaahh, ok, yes, that's a good idea.

rgmf, 2020-04-14:
When you export a Track without TrackPoints this is the kml (for example) generated:
 Look at this kml file that corresponding to a 1 hour activity without TrackPoints:

```kml
<TimeStamp><when>1970-01-01T00:00:00Z</when></TimeStamp>
<styleUrl>#start</styleUrl>
<Point>
<coordinates>0.0,0.0</coordinates>
</Point>
...
<TimeStamp><when>1970-01-01T00:00:00Z</when></TimeStamp>
<styleUrl>#start</styleUrl>
<Point>
<coordinates>0.0,0.0</coordinates>
</Point>
```

That's why you get a bad Track when you import this files (bad date and bad total time).

rgmf, 2020-04-15:
Max and Min Elevation in Track description have weird values. I think this values are `Double.POSITIVE_INFINITY` and `Double.POSITIVE_INFINITY`.

```kml
Max elevation: -9223372036854775808 m (-9223372036854775808 ft)
Min elevation: 9223372036854775807 m (9223372036854775807 ft)
```

Ideas:
- max elevation == Double.POSITIVE_INFINITY means that there isn't max elevation information --> Write down in description some descriptive message (like "Unknown") or 0 or don't write down this information ¿?
- min elevation == Double.POSITIVE_INFINITY means there isn't min elevation information --> Write down in description some descriptive message (like "Unknown") or 0 or don't write down this information ¿?

@dennisguse @pstorch What do you think?

dennisguse, 2020-04-15:
@rgmf I guess this is a result of `TrackStatistics` - what about adding something like this?
```
public boolean hasElevationMax() {
  return Double.isInfinite(elevationMax);
}
```

closed 2020-04-23 by rgmf