OpenTracksApp/OpenTracks #181: heart rate export issue

Pilou8, 2020-04-11:
Hello,

OpenTracks was initaly my prefered app to monitor and record Heart Rate when biking.
But I have not been able to export hr data and import them in MyTourBook for example.

If I use AAT for recording, I can then export a .gpx file containing HR data and import it in MyTourBook.
But if I export .gpx from OpenTracks, I get no HR data in the file.
And if I export a .kmz / .kml file frome OpenTracks , I get a list of HR data at the end of the file, but I can't figure out how they can be related to time or position and I found no software able to read theses HR data (only geo data).

Would there be a solution to export HR data towards other software, ideally through a .gpx file like AAT or RunnerUp do?

Thanks a lot for your work and attention !

dennisguse, 2020-04-11:
GPX by default does not support heartrate (or IMHO any other sensor data), but there are some extensions that could be used.

Strava has a good overview of they support for GPX: https://developers.strava.com/docs/uploads/

Pilou8, 2020-04-12:
<span style="font-family:arial,helvetica,sans-serif; font-size:12px"> </span><span style="font-family:arial,helvetica,sans-serif; font-size:12px"> Thanks for your answer!<br>
<br>
Interesting link actually!<br>
I understand that heartate is not officially supported by gpx standard.<br>
<br>
But could it be possible for OpenTracks to export gpx with heart rate using a format that would simultaneously be:<br>
- supported by MyTourBook, GPXSee, Viking, etc.<br>
- the same as exported by AAT or RunnerUp (which a supported by MyTourBook)<br>
- probably the most common among the 3 supported by by Strava<br>
<br>
Thanks again!</span><br>
<br>
<br>
<br>
<br>

<div class="gl_quote" style="margin-top: 20px; padding-top: 5px;">De : "Dennis Guse"<br>
A : "OpenTracksApp/OpenTracks" ,"Pilou8" ,"Author"<br>
Envoyé: samedi 11 Avril 2020 20:07<br>
Objet : Re: [OpenTracksApp/OpenTracks] GPX: export and import heartrate data (#181)<br>
 
<div class="gl_quoted">
<p> </p>

<p>GPX by default does not support heartrate (or IMHO any other sensor data), but there are some extensions that could be used.</p>

<p>Strava has a good overview of they support for GPX: <a href="https://developers.strava.com/docs/uploads/" rel="nofollow">https://developers.strava.com/docs/uploads/</a></p>

<p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
</div>

<div class="gl_quoted"> </div>
</div>

MrS1R1US, 2020-04-14:
I support the author of the message! The heart rate sensor is very important in sports and very important in exporting from the application.

daloop, 2020-05-16:
Import of embedded heart rate data would be nice, indeed.

I use Open Tracks as a GPX viewer in my phone and I import gpx files that have heart rate data embedded via gpx extensions.

I use data from a TomTom Spark Cardio and convert it to gpx with [ttwatch](https://github.com/ryanbinns/ttwatch)
The format is:
<trkpt lon="x.x" lat="x.x">
                <time>2020-03-19T13:45:25.000Z</time>
                <extensions>
                    <gpxtpx:TrackPointExtension>
                        <gpxtpx:hr>75</gpxtpx:hr>
                    </gpxtpx:TrackPointExtension>
                </extensions>
            </trkpt>

ronajon, 2020-05-18:
GPX with heart rate would be very nice.
another route could be to add exporting to TCX files. 
These have default heart rate in their scheme:
https://www8.garmin.com/xmlschemas/TrainingCenterDatabasev2.xsd

dennisguse, 2020-05-18:
@ronajon TCX was already implemented in MyTracks, but I removed due to maintenance reasons
It already implemented sensor data (heart rate, power etc.)

BUT the code base severely changed since then, so it there is some work need to polish everything.

https://github.com/OpenTracksApp/OpenTracks/commit/8255fef72db193d90b2ebbca945e2c9d8165d099

commit 82e0863f (2020-07-24): GPX import/export: speed via TrackPointExtension/v2.
Fixed data type for heartrate/speed (no decimals) allowed.
Fixes #335.
Fixes #181.

GPXTrackWriter: set Trackpointv2 header correctly and export time in correct sequence.
PR #337 (2020-07-24): GPX: export heartrate, cadence, and speed via TrackPointExtension/v2
This is a breaking change as speed exported with v3.7.6 cannot be imported.

Fixes #181.
Follow-up to #336.

@ronajon i fixed some smaller things (added missing members to AbstractFileImporter, enabled the tests, and changed speed to also use TrackPointExtension/v2.)
Well prepared :+1: 
It was just easier for to create a branch in the main repo...
commit 065f42fb (2020-07-24): GPX import/export: speed via TrackPointExtension/v2.
Fixed data type for heartrate/speed (no decimals) allowed.
Fixes #335.
Fixes #181.

GPXTrackWriter: set Trackpointv2 header correctly and export time in correct sequence.
closed 2020-07-24 by dennisguse