OpenTracksApp/OpenTracks #185: Preserving the heart rate history of the entire workout and pulse zones

MrS1R1US, 2020-04-14:
** Is your feature request related to a problem? Please describe. **
Heart rate analysis by time or distance of training is lacking as it is implemented in Polar Beat or Runtastic applications.

** Describe the solution you'd like **
Add custom heart rate zones and a historical heart rate graph by time or distance training. For an athlete this is very important.

** Describe alternatives you've considered **
Add custom heart rate zones and a historical heart rate graph by time or distance training. For an athlete this is very important.

** Additional context **
Screenshots:
[https://i.hizliresim.com/ZXnmp3.jpg](url)
[https://www.wareable.com/media/imager/201606/14064-original.jpg](url)

MrS1R1US, 2020-04-19:
What can you say about this? Does it make sense to wait for such an introduction? Very critical is not enough.

dennisguse, 2020-04-19:
@MrS1R1US It definitively makes sense to wait :)
If you have some time, you could create a concept how to implement this (e.g., screen shots / scribbles).

One (technical) issue might be that only _one_ heart rate measurement is stored per GPS location.
So, if a GPS is recorded every 10 seconds the only _last_ measurement in this period is stored.
For longer intervals this might even be more problematic...

MrS1R1US, 2020-04-19:
@dennisguse 
If you really do this, I will certainly draw a layout in Photoshop about how it will look. And I will be very grateful to you for such functionality!) An interval of 10 seconds for large distances is normal. For small workouts, 5-10 minutes is not very. In any case, you need to try, because this will attract the attention of athletes who train with sensors like me. For the sake of OpenTracks, I demolished the official firmware and installed Lineage OS, after which I got GPS.

DevRonin, 2020-04-30:
What phone model do you have?
I also have LineageOS 15 (Android 8.1.0) on LeEco LE2 snapdragon version.
I ordered a Polar H10 and Garmin HRM dual.

As a starter, a new tab, with a  simple graph like the one with elevation vs distance will be good.
Y - heart rate
X - this could be time, distance, speed, etc.

Beside current HR, some stats for HR training will be good, like total time spent on certain interval, max. continuous time in that interval, etc.
Also possibility to set a target HR interval & low(under)/high(over) limit alarm will be nice.

DevRonin, 2020-04-30:
> @MrS1R1US It definitively makes sense to wait :)
> If you have some time, you could create a concept how to implement this (e.g., screen shots / scribbles).
>
> One (technical) issue might be that only [...]
Separate log for HR also? together maybe with other sensors data (cadence, speed) ?

MrS1R1US, 2022-02-05:
> What phone model do you have? I also have LineageOS 15 (Android 8.1.0) on LeEco LE2 snapdragon version. I ordered a Polar H10 and Garmin HRM dual.
>
> As a starter, a new tab, with a simple graph like [...]

Describe the solution you'd like
There is a very lack of heart rate zones, as in Polar Beat. When the heart rate zones are set and there is a real-time color graph, it is easier to follow the correct heart rate. This is a very important schedule during a serious workout.

Additional context
This is what the final chart looks like. Also, the training graph with a line is built heart rate, which allows you to see the line that you cannot cross.
[image]

This is how the graph looks like during a workout, the line moves depending on the heart rate zone. Over time, when the workout is long, the first heart rate point goes beyond the graph in order to make the graph smoother. Please note that the heart rate numbers themselves change color depending on the heart rate zone, this is very helpful for real-time analysis to slow down or increase it.

This is how I would like to see it in OpenTracks:
[image]

This is a very important function, to see in which pulse zone you are. This feature is available in all sports devices to which a heart rate monitor is connected. In the settings you set the maximum heart rate, and then it is calculated down from 100%. It is also worth noting that it is important to save the graph at the end of the workout, this will allow you to compare with past changes.
Zone 5 - 90-100%
Zone 4 - 80-90%
Zone 3 - 70-80%
Zone 2 - 60-70%
Zone 1 - 50-60%

This is how the final heart rate graph of the finished workout looks like:
[image]