OpenTracksApp/OpenTracks #186: How can I make a permanent automatic backup?

MrS1R1US, 2020-04-14:
Good afternoon. Tell me, I want make a permanent automatic backup. Where are training files stored to automatically duplicate them on my computer? Clicking "export" in the settings is a big time consuming.

dennisguse, 2020-04-14:
The data is stored in a SQLite database.
So, there are no files until "Export all" is triggered.
@MrS1R1US does this answer your question?

MrS1R1US, 2020-04-14:
> 
> 
> The data is stored in a SQLite database.
> So, there are no files until "Export all" is triggered.
> @MrS1R1US does this answer your question?

Can I synchronize my workouts using SQLite database synchronization?

dennisguse, 2020-04-14:
For backup purpose: yes, for synchronization: no.
I also would like to have a synchronization feature, but it is rather hard to implement and so far there have always other more important things (e.g., #10).

pstorch, 2020-04-14:
If the database is up to 25 MB and the app was installed via Google PlayStore, we could enable the autobackup feature of Android:
https://developer.android.com/guide/topics/data/autobackup

https://github.com/OpenTracksApp/OpenTracks/blob/b09eeaede342f8524e56de91019a6928996ebbf3/src/main/AndroidManifest.xml#L22

But I guess a backup is mostly requested by power users which have lots of tracks (and photo markers), then the 25 MB limit is easily crossed.

MrS1R1US, 2020-04-14:
> 
> 
> For backup purpose: yes, for synchronization: no.
> I also would like to have a synchronization feature, but it is rather hard to implement and so far there have always other more important things (e.g., #10).

In case of reinstalling the application, can I recover the data?

How do I get to this base, where does it lie? I will connect it through syncthing(app) to the PC

MrS1R1US, 2020-04-14:
> If the database is up to 25 MB and the app was installed via Google PlayStore, we could enable the autobackup feature of Android

No. F-droid.

pstorch, 2020-04-14:
Via the Storage Access Framework we could let the user choose a backup folder and copy the SQL-Lite DB there after each new Track was recorded for example. This backup directory could be one for syncthing or nextcloud, or whatever is available via SAF.

MrS1R1US, 2020-04-14:
> 
> 
> Via the Storage Access Framework we could let the user choose a backup folder and copy the SQL-Lite DB there after each new Track was recorded for example. This backup directory could be one for syncthing or nextcloud, or whatever is available via SAF.

Great) How do I use the Storage Access Framework?

pstorch, 2020-04-14:
> Great) How do I use the Storage Access

This would need to be implemented in the app.

MrS1R1US, 2020-04-14:
> 
> 
> > Great) How do I use the Storage Access
> 
> This would need to be implemented in the app. And a restore from this directory as well.

How do I use the Storage Access Framework?

Then I can import this database in case of what?

pstorch, 2020-04-14:
It time permits, I might be able to work on a backup/restore feature.

dennisguse, 2020-04-14:
About a backup and restore functionality: I would prefer to stick with the KMZ approach.
First of all, it is independent of the OpenTracks version (at least somehow).
And also it contains the pictures (that are not stored in the SQLite database).

I agree that the current implementation is not user-friendly for backups (too many clicks required).
However, I would not like to implement an automatic approach that dumps the data onto external storage as it is not privacy-friendly.
At least no until Android enforcing the use of SAF.

What we could implement is something like a ContentProvider/DocumentProvider.
However, I have no clue if this is actually possible.
What I sometimes would require is the option to synchronize OpenTracks via NextCloud (see #90).

dennisguse, 2020-04-14:
And about `allowBackup=true`: I am not a big fan of sending backups to Google's infrastructure.
Also it would not be possible to backup the images (probably too large).

pstorch, 2020-04-14:
No, we don't need to discuss the Google Backup any further.

Is the KMZ export containing really all data? I'm not familiar with it at the moment.

Regarding the privacy friendly external automatic backup: if the user chooses the directory (via SAF) then it it his/her choice. I don't see a privacy issue here. And as I wrote, this directory could be a nextcloud directory (at least this is offered me, when I choose a directory via SAF).

dennisguse, 2020-04-14:
@pstorch You are right - giving the user the choice is preferable.
At the moment, the KMZ files are almost complete - there are some (smaller) issues that the statistics are not completely identical.
And there might be some bugs, but for now it should almost result in a perfect copy.

rgmf, 2020-04-14:
> @pstorch You are right - giving the user the choice is preferable.
> At the moment, the KMZ files are almost complete - **there are some (smaller) issues** that the statistics are not completely identical.

@pstorch @dennisguse this afternoon testing import/export I noticed that there is an issue with Tracks without points. In this case the only data you have is total time but when you import them and export them this total time is lost.

Did you know it? Is this one of the small issues you tell us? Do you consider it like a issue or it's a normal behavior?

dennisguse, 2020-04-14:
@rgmf I guess that is related - if I got it right, the TrackStatistics are recreated on import using the exported TrackPoints except that the startTime is not exported at all.
In addition, the stop time might also not be exported as no TrackPoint is created then.
The exported summary is not parsed.

rgmf, 2020-04-14:
@dennisguse yes, that seem.

MrS1R1US, 2020-04-19:
> About a backup and restore functionality: I would prefer to stick with the KMZ approach.
> First of all, it is independent of the OpenTracks version (at least somehow).
> And also it contains the [...]

I would like to receive the necessary file in the internal storage of the phone, and already using "syncthing" app to synchronize with my personal PC - it's safe.

MrS1R1US, 2022-02-04:
Thank you very match!!!:)

LuCarhub, 2023-06-02:
Hi there, 
can someone let me know in what folder opentracks saves the pictures I took? 
How can I know trhe structure of the directory used to save info (where is database, photo, etc.)

pstorch, 2023-06-02:
It's in the internal app-private storage. Usually you cannot access it. Only with a rooted phone and root enabled file manager. Or with a ADB when your phone is connected to a PC in debug mode, I think.

LuCarhub, 2023-06-02:
Il 02/06/2023 22:24, Peter Storch ha scritto:
>
> It's in the internal app-private storage. Usually you cannot access 
> it. Only with a rooted phone and root enabled file manager. Or with a 
> ADB when your phone is connected to a PC in debug mode, I think.
>
Yes, I have a rooted phone. I can make a backup with external data using 
titanium backup, but I want to understand whre are my pictures, see the 
size of them, etc. So, can you say me the exact path?

pstorch, 2023-06-03:
On my device I found the database here:
`/data/data/de.dennisguse.opentracks(.debug|.nightly|.playstore)/databases`
This folder contains the files `database.db` and `database.db-journal`.

and the pictures here:
`/storage/self/primary/Android/data/de.dennisguse.opentracks(.debug|.nightly|.playstore)/files/Pictures`

The `Pictures` folder has subdirectories per TrackId.

`(.debug|.nightly|.playstore)` depending on the version you've installed.

dennisguse, 2023-06-03:
@pstorch Should we document this somewhere?

pstorch, 2023-06-03:
@dennisguse good question. I'm not sure if parts of the paths are device or android version specific.

LuCarhub, 2023-06-03:
Can I suggest if it's possible to use directory that is possible to backup
also for non rooted phone? Anyway, I have a rooted phone and the first app
I've tried (alpha backup pro, not  available anymore on play store) didn't
backup everything (picture lost). Later I'll try with swift backup.

Honestly the problem of backup is a big issue, i can't imagine how long I
can take to export and import years of activities. It's much easier to
backup the app.

pstorch, 2023-06-03:
The automatic post workout export function can be used to automatically backup your workouts to a different place (other directory or cloud storage via supported app).

LuCarhub, 2023-06-04:
Could it be possible to store database in a specific folder? And also pictures, maybe, under /pictures? At the moment it's a hard to create a backup for me.

My experience on Samsung S20 Ultra 5G: 
I installed opentracks from f-droid.

Pictures are in: /storage/self/primary/Android/data/de.dennisguse.opentracks/files/Pictures/...

I didn't find database in /data/data/de.dennisguse.opentracks... I'm still looking for it, I tried to backup (using a plugin of magisk) the entire /data of my phone and I've searched on the log of backup. There is only:
/data/overlays/preferences/ThemeDatabase.db
/data/overlays/preferences/ThemeDatabase.db-journal
related to database.db, so the database has a different name. The other folder and files under "dennisguse" are:

/data/app/~~PDjxxllZaQk9h6cRehCacw==/de.dennisguse.opentracks--h7tu6ZkpanEiZCDeF7ktg==/
/data/app/~~PDjxxllZaQk9h6cRehCacw==/de.dennisguse.opentracks--h7tu6ZkpanEiZCDeF7ktg==/oat/
/data/app/~~PDjxxllZaQk9h6cRehCacw==/de.dennisguse.opentracks--h7tu6ZkpanEiZCDeF7ktg==/oat/arm64/
/data/app/~~PDjxxllZaQk9h6cRehCacw==/de.dennisguse.opentracks--h7tu6ZkpanEiZCDeF7ktg==/base.apk
/data/app/~~PDjxxllZaQk9h6cRehCacw==/de.dennisguse.opentracks--h7tu6ZkpanEiZCDeF7ktg==/base.dm
this are part of application

/data/system/graphicsstats/1684627200000/de.dennisguse.opentracks/5211/total
this file, "total", is just 1,2KB

same results searching for "opentracks". I'm having trouble examining the backup created using swiftbackup, one of them is password protected. I have to try to restore it on another phone, I'll let you know.

LuCarhub, 2023-06-04:
Works! Backup made with Swiftbackup (free) on a rooted phone works! I created a full backup of app, with all external files, and I restored it on another phone, and I saw activities, statistics and pictures.
Anyway, I don't think it's possible to to the same on a non-rooted phone.

pstorch, 2023-06-04:
Why not utilize the built in automatic export function as backup?

LuCarhub, 2023-06-04:
Hi pstorch,

just because a massive export/import could take more time than a easy full backup. Consider, for example, if you change phone after some years. I think there's a big export/import to do. And, with a full backup of app, you can save also personalization of app.
This is my opinion, maybe it's not so important for other users.

dennisguse, 2023-06-04:
@LuCarhub let me put it this way: pull request welcome.

What we support so far:
* export/import one file for all tracks or one track per file
* ADB backup is enabled
* Android's device to device transfer might be working: https://github.com/OpenTracksApp/OpenTracks/issues/1110

If you can figure out a better way how to implement this feel free to contribute.

LuCarhub, 2023-06-04:
Dennis,

I'm totally satisfied by your reply. Thank you.
Thank you also for all the support you're providing.

closed 2023-06-07 by dennisguse

LuCarhub, 2024-02-18:
> Works! Backup made with Swiftbackup (free) on a rooted phone works! I created a full backup of app, with all external files, and I restored it on another phone, and I saw activities, statistics and pictures. Anyway, I don't think it's possible to to the same on a non-rooted phone.

I tried today to restore an old backup, sadly I have all tracks but pictures was lost. So the best way is to export everything.

dennisguse, 2024-02-19:
@LuCarhub did you try to restore to the same OpenTracks version or a newer one?