OpenTracksApp/OpenTracks #189: Should marker's photos be deleted when track is delete?

rgmf, 2020-04-18:
**Is your feature request related to a problem? Please describe.**
I think so.
When you delete tracks with markers that has photos, the track is deleted but images remain in the external storage. Should they be deleted?

**Reproduce:**
1.- Record a new track, add marker with photos. These photos will be saved in OpenTracks external storage.
2.- Stop recording.
3.- Delete this track.
4.- Open OpenTracks external storage with a File Explorer and you'll see the photos added in the step 1.

dennisguse, 2020-04-19:
@rgmf You are right: and that is kind of a nasty bug as it blocks storage on the user's devices.
Are you looking into it?
Could you create a test for this?

rgmf, 2020-04-19:
> Are you looking into it?
Yes, I am.

> Could you create a test for this?
Of course ;)

commit 240d8f6b (2020-04-27): Images/folders are deleted when tracks/waypoints are deleted.

This fixes #189.
closed 2020-04-25 by rgmf